mne-tools/mne-python #2433: Simulating EEG/MEG/etc Data?

choldgraf, 2015-08-25:
Hey all - I see that MNE has a simulations module, but it's pretty restrictive right now. Are there plans to include support for simulating other kinds of signals? Or is there another package that already allows you to do this relatively easily?

In particular, I'm looking to simulate 1/f noise, add some event-locked power increases in particular frequency bands, and then try out a few preprocessing techniques to see which results in the cleanest output data. I was gonna code this up myself but figure it's something that has probably already been done. Any ideas?

Eric89GXL, 2015-08-25:
1/f is already supported using the iir parameters in the `generate_noise` functions that e.g. `generate_evoked` uses

Right now you can only simulate evoked data, but there is a PR #2295 to do it with `Raw` data. I probably need to take that one over and get it in soon. I think that would give you the flexibility you want.

The only supported datatypes right now are MEG and EEG. The limitation comes from the forward solution calculation code, so if we want to simulate other datatypes we probably need to beef up our forward code. Otherwise we'll have to do a bit of refactoring of the WIP `Raw` simulation to allow passing precomputed forwards, but it wouldn't be too terrible I don't think.

choldgraf, 2015-08-25:
Cool - right now I'm not worried about creating something that's super biologically realistic (maybe I should be)...just simulating noise and adding some time-locked components. Besides I'm in the world of ecog, so who knows what kind of forward solution we've got compared w/ EEG.

Eric89GXL, 2015-11-25:
@choldgraf I think you can do what you want already with `simulate_raw`, no?

choldgraf, 2015-11-25:
Looks like that would work well - it is overkill for me because I don't have anything regarding a source, anatomical information, etc. (since I'm doing ECoG and not MEG/EEG). I just wanted an easy way to generate 1/f noise and add some frequency components that are stimulus-responsive.

Eric89GXL, 2015-11-25:
The 1/f noise (with a covariance of your choice) you can get from the simulation module. Adding some frequency components sounds like adding `np.sin` with appropriate args. Is that enough? Do you have some mne-python API in mind? I'm not sure if this use case will be generally useful enough to warrant its own function, but an example could be nice.

choldgraf, 2015-11-25:
Nah I think you're probably right in that what's there now is fine. If
something else pops into my head then I'll mention it here, but I think we
can close this for now.

closed 2015-11-25 by choldgraf