mne-tools/mne-python #2434: EpochsArray constructor is broken if len(times) == 1

dengemann, 2015-08-26:
This is useful for EEG/MEG measures which reduce the time dimension in each trial. It used to work with the last release. Tried to find a fix but it seems to be deeper than I assumed.

cc @Eric89GXL @agramfort

dengemann, 2015-08-26:
we cuttently have this one in epochs.py:

https://github.com/mne-tools/mne-python/blob/master/mne/epochs.py#L277

It won't play nicely if shape[-1] == 1

as round((tmax - tmin) can be 0

dengemann, 2015-08-26:
cc @fraimondo

dengemann, 2015-08-26:
sorry parsed the code incorrectly, so if only 1 time sample exists, the conditional breaks.

agramfort, 2015-08-26:
can you share a gist that reproduces the bug?

agramfort, 2015-08-26:
provide a tiny test and we'll make it pass

dengemann, 2015-08-26:
> provide a tiny test and we'll make it pass

How about this PR instead ? #2435 ;)

PR #2438 (2015-08-27): FIX: Allow one-sample epochs
cc @dengemann

Closes #2435
Closes #2434

~~Can you make a minimal example for the other error you were getting having to do with `event_id` being an integer?~~
closed 2015-08-27 by dengemann