mne-tools/mne-python #2439: auto generate references

mainakjas, 2015-08-28:
We should use this: https://sphinxcontrib-bibtex.readthedocs.org/en/latest/quickstart.html in the docstrings to auto generate the reference page: http://martinos.org/mne/dev/references.html. It may need some hacking to get the cross-links working between the docstrings and this page. Thoughts?

agramfort, 2015-08-29:
+1 as discussed IRL

can you paste how a new docstring would look like?

mainakjas, 2015-09-01:
Here you go. Instead of 

```
 References
 ----------
    [1] Zoltan J. Koles. The quantitative extraction and topographic mapping
    of the abnormal components in the clinical EEG. Electroencephalography
    and Clinical Neurophysiology, 79(6):440--447, December 1991.
```
you would have

```
.. bibliography:: refs.bib
   :list: bullet
   :filter: author % "Koles"
```

and `refs.bib` contains among other entries, this one:
```py
@article{koles1991quantitative,
  title={The quantitative extraction and topographic mapping of the abnormal components in the clinical EEG},
  author={Koles, Zoltan Joseph},
  journal={Electroencephalography and clinical Neurophysiology},
  volume={79},
  number={6},
  pages={440--447},
  year={1991},
  publisher={Elsevier}
}
```

Eric89GXL, 2015-09-01:
On the one hand this is a bit nicer because we can keep our refs in one place, on the other hand it actually makes the plain-text documentation far less useful than it was before. So I'm -1 on this currently. Perhaps I'm missing the motivation that overcomes this drawback, though. Do we want/need all our refs in one place? Does it help to keep them in the standard .bib format? Do we repeat a lot of references and this will be more DRY?

mainakjas, 2015-09-01:
woops. You need a slight modification I think.
```
:cite:`1991:koles1991quantitative` 
.. bibliography:: refs.bib
   :filter: author % "Koles"
```
And in the `references.rst` page (this time no filters), you'll have:

```
:cite:`1991:koles1991quantitative`
.. bibliography:: refs.bib
```
This is closer to what we want. We'll get cross-refs then between function documentation and the references.html page. The current approach doesn't create cross-refs. Sorry, got buried in the details :)

agramfort, 2015-09-01:
I also don't like the fact that the plain ref is not readable in the text
anymore. Is the way around this to duplicate it with :cite: + plain text?

mainakjas, 2015-09-01:
Indeed, that's bad if you're looking at the doc from ipython or so. Yes, you'll have to duplicate the plain text then. We need automatic parsers if you don't want to duplicate. Duecredit is going to [work on similar lines](https://github.com/duecredit/duecredit/blob/master/duecredit/parsers.py).

agramfort, 2015-09-03:
I guess it should be done there.

ask @yarikoptic on what he would need help

yarikoptic, 2015-09-03:
indeed it would be awesome to get citations for mne-python functionality/methods in duecredit.  There are two ways to do it ATM: 
https://github.com/duecredit/duecredit/#native-use-of-duecredit-recommended
If you are keen on adopting a "native way", I can initiate a PR.

So far though I have tried to stay from any "integration" although it could be done (e.g. duecredit enhancing docstrings with references from its own entries), but then it would demand strict dependency on duecredit, which might be a slight burden.  May be it somehow could be implemented at "build time" though (duecredit "maintaining" References within docstrings in one form or another and collating target .bib) and thus avoiding run-time dependency. It somewhat also relates to my own use-case of simplifying collection of "use" references for a toolkit: https://github.com/duecredit/duecredit/issues/7 since as a result it should generate e.g. the .rst to be included in the documentation (and replace manually built ones such as http://www.pymvpa.org/whoisusingit.html#studies-employing-pymvpa .
But that sounds like a good piece of work TODO.  On the other hand, we could make duecredit use references within your docstrings, but then we would need to enhance them with additional information since duecredit supports meta-information about every citation, such as a brief description of the functionality/purpose, tags (either it is a reference implementation of the method, or mere "educational" reference, etc).

mainakjas, 2015-09-03:
@yarikoptic just to set up the context a bit. I'm exploring this area with
@agramfort to figure out if there's something useful and impactful that
could be done. This could _potentially_ translate into one month of
full-time work spread over one year. Maybe we could team up :)

> indeed it would be awesome to get citations for mne-python
> functionality/methods in duecredit. There are two ways to do it ATM:
> https://github.com/duecredit/duecredit/#native-use-of-duecredit-recommended
> If you are keen on adopting a "native way", I can initiate a PR.
I think @agramfort can comment on this. I'm curious to see how this is
supposed to be configured though. Just following the instructions on the
duecredit site didn't seem to work for me. I get this
<https://gist.github.com/mainakjas/38b92b0c284242e91e72> error.

> So far though I have tried to stay from any "integration" although it
> could be done (e.g. duecredit enhancing docstrings with references from its
> own entries), but then it would demand strict dependency on duecredit,
> which might be a slight burden. May be it somehow could be implemented at
> "build time" though (duecredit "maintaining" References within docstrings
> in one form or another and collating target .bib) and thus avoiding
> run-time dependency. It somewhat also relates to my own use-case of
> simplifying collection of "use" references for a toolkit:
> duecredit/duecredit#7 <https://github.com/duecredit/duecredit/issues/7>
> since as a result it should generate e.g. the .rst to be included in the
> documentation (and replace manually built ones such as http://www.pymvpa.org/whoisusingit.html#studies-employing-py
> mvpa <http://www.pymvpa.org/whoisusingit.html#studies-employing-pymvpa> .
Indeed, this is exactly what we want. Auto-generate this page. But also
with cross-refs back to the functions which implement it. Consider it to be
an interactive alternative complementary to the command-line utility :)
Except I'm not sure if we want to add the references manually ("injection"
and "native" both appear manual to me). Automatic could scale better (of
course with caveats). You could another option called "automatic" to these
two existing approaches.

> But that sounds like a good piece of work TODO.
That's indeed the idea ;)

> On the other hand, we could make duecredit use references within your
> docstrings,
Yep, that's what we want to do :)

> but then we would need to enhance them with additional information since
> duecredit supports meta-information about every citation, such as a brief
> description of the functionality/purpose, tags (either it is a reference
> implementation of the method, or mere "educational" reference, etc).
We could use the first line of the docstring for the brief description. And
the tag could be "unknown" or guessed from other available clues?

yarikoptic, 2015-09-04:
On Thu, 03 Sep 2015, mainakjas wrote:
> ... full-time work spread over one year. Maybe we could team up :)

"team up" sounds awesome!   Not sure if I would be able to contribute
as much of my time though ;)

> > indeed it would be awesome to get citations for mne-python
> > functionality/methods in duecredit. There are two ways to do it ATM:
> > https://github.com/duecredit/duecredit/#native-use-of-duecredit-recommended
> > If you are keen on adopting a "native way", I can initiate a PR.

> I think @agramfort can comment on this. I'm curious to see how this is
> supposed to be configured though. Just following the instructions on the
> duecredit site didn't seem to work for me. I get this
> <https://gist.github.com/mainakjas/38b92b0c284242e91e72> error.

wow -- that is quite a blow!  I guess I do need to learn how to use anaconda
and to replicate this issue.  I believe Matteo (duecredit co-author) uses it
but we never saw such a crash.  duecredit is quite new and despite being
developed under linux and osx, there could still be surprises ;)

Just for completeness -- that 'report.py' -- is that something you
could share?  Any additional detail on how to replicate would be
appreciated!  (there is now 0.4.2 of duecredit, but I don't think it
resolved any related issue)

> > duecredit/duecredit#7 <https://github.com/duecredit/duecredit/issues/7>
> > since as a result it should generate e.g. the .rst to be included in the
> > documentation (and replace manually built ones such as http://www.pymvpa.org/
> whoisusingit.html#studies-employing-py
> > mvpa <http://www.pymvpa.org/whoisusingit.html#studies-employing-pymvpa> .
> Indeed, this is exactly what we want. Auto-generate this page. But also
> with cross-refs back to the functions which implement it.

that page is just of the "users" of PyMVPA, so not exactly aiming to point to
functionality they have used, but eventually we want to provide such as
"service" ;) 

In your case you want to collate references for all the functionality
implemented in MNE.  Indeed it is a nice use case, inline with the
"users" one (e.g.  just output list of citations with "implementation"
instead of "use" tag), and then back-references indeed would be meaningful.
Added a comment to that issue #7 of duecredit ;)

> Consider it to be an interactive alternative complementary to the
> command-line utility :) Except I'm not sure if we want to add the
> references manually ("injection" and "native" both appear manual to
> me). 

well -- somewhere reference needs to be added, right? ;)  if you get a
centralized .bib, you could just

due.load('whatever.bib') somewhere e.g. in mne/__init__.py (would do nothing
unless duecredit is "Active")  and then use bibtex keys in due.dcite /
due.cite calls.

> Automatic could scale better (of course with caveats). You
> could another option called "automatic" to these two existing
> approaches.

with 'automatic' you mean scraping of the docstrings?

With duecredit there is an additional feature/caveat.  The idea is to collect
references for functionality which is actually used.  So, if there is a
reference within class docstring, and someone might even initiate that object
but don't really use it (i.e. to call some "representative" method), we
would prefer not to cite that reference in the report.

Overall -- some 'manual' specification I guess would be necessary to
denote running of which functionality constitutes "using" the referenced
method/functionality.

> > On the other hand, we could make duecredit use references within your
> > docstrings,

> Yep, that's what we want to do :)

so the simplest/non-ambiguous way I see is smth like

```
class CSP(TransformerMixin):
   """....
   :cite:`1991:koles1991quantitative` 
   .. bibliography:: refs.bib
   """
   ...
   @due.dcite(due.USE_CLASS_DOC)
   def fit(...
```

where we would implement support for delayed (since at decorator creation time
there is no access to the class's __doc__ AFAIK) parsing of
self.__class__.__doc__ whenever that fit gets called.  "implementation" would
be assumed to be the default tag (otherwise could be specified in dcite call),
and description would come from the first line in the __doc__.

Or in injections approach we could support specifying from which docstring to
scrape the reference (if machine parseable like with :cite:)

Smth like that?

> > but then we would need to enhance them with additional information since
> > duecredit supports meta-information about every citation, such as a brief
> > description of the functionality/purpose, tags (either it is a reference
> > implementation of the method, or mere "educational" reference, etc).

> We could use the first line of the docstring for the brief description. And
> the tag could be "unknown" or guessed from other available clues?

description -- agreed.  tags -- we can only assume default (e.g.  "reference"),
not sure if could be somehow guessed, unless explicitly provided somewhere (as
an additional attribute in :cite: or in @due.dcite arguments)

-- 
Yaroslav O. Halchenko
Center for Open Neuroscience     http://centerforopenneuroscience.org
Dartmouth College, 419 Moore Hall, Hinman Box 6207, Hanover, NH 03755
Phone: +1 (603) 646-9834                       Fax: +1 (603) 646-1419
WWW:   http://www.linkedin.com/in/yarik

yarikoptic, 2015-09-04:
Re
```
.. bibliography:: refs.bib
   :filter: author % "Koles"
```
gy -- finally was done properly? ;)  Since awhile for our lab we use my hack (https://github.com/matthew-brett/bibstuff/pull/7, never got to finish up) on top of Matthew's craft on top of abandoned bibstuff which looks very similar:
```
.. bibmatching:: ../_static/haxbylab.bib
    :style: haxby
    :filter:
     author: \bHalchenko\b
    :groupby: entry_type, subtype
    :sort:
```
which results in http://haxbylab.dartmouth.edu/ppl/yarik.html#publications

mainakjas, 2015-09-04:
>
> "team up" sounds awesome! Not sure if I would be able to contribute
> as much of my time though ;)
>

let's see how things go ... :)

> Just following the instructions on the
>
> wow -- that is quite a blow! I guess I do need to learn how to use anaconda
> and to replicate this issue. I believe Matteo (duecredit co-author) uses it
> but we [...]

report.py is in mne-python itself. To reproduce, just do the following:

$ git clone https://github.com/mne-tools/mne-python.git && cd
mne-python/mne/
$ wget -q -O ./due.py
https://raw.githubusercontent.com/duecredit/duecredit/master/duecredit/stub.py
$ easy_install duecredit # use the anaconda easy_install
$ python -m duecredit report.py

not sure if this is how it's supposed to be used though ... I didn't need
to insert any 'native' Due citations to get the error.

> In your case you want to collate references for all the functionality
> implemented in MNE. Indeed it is a nice use case, inline with the
> "users" one (e.g. just output list of citations with [...]

I like that. Why not.

> > Consider it to be an interactive alternative complementary to the
> > command-line utility :) Except I'm not sure if we want to add the
> > references manually ("injection" and "native" both appear manual to
> > me).
>
> well -- somewhere reference needs to be added, right? ;)

indeed, my only complain is that you are "changing habits". Imagine having
to convert all the existing plaintext references to bibtex ...

> > Automatic could scale better (of course with caveats). You
> > could another option called "automatic" to these two existing
> > approaches.
>
> with 'automatic' you mean scraping of the docstrings?
>

yep

> With duecredit there is an additional feature/caveat. The idea is to
> collect
> references for functionality which is actually used. So, if there is a
> reference within class docstring, and someone [...]

makes sense. I guess it's a complementary use case. In our case, we are
looking to generate a bibliography for all the functions in the package
without any filters in terms of usage. Maybe later (and I am digressing),
it would be neat to generate an "html report" which will list all the
references used (like in the command line utility) but also offer clickable
links to specific parts of the code.

Overall -- some 'manual' specification I guess would be necessary to
> denote running of which functionality constitutes "using" the referenced
> method/functionality.
>
>
>
> so the simplest/non-ambiguous way I see is smth like
>
> ```
> class CSP(TransformerMixin):
> """.... [...]

so you would see a plaintext documentation when you do csp.fit? in ipython?
Or in the html generated using sphinx? But you can do this only for methods
of a class, not the class itself? sounds a bit hackish but if it works, why
not.

> Or in injections approach we could support specifying from which docstring
> to
> scrape the reference (if machine parseable like with :cite:)
>
> Smth like that?
>

because you want to list only "some" of the references -- the ones that are
actually "used"? This brings me to another question. I understand that you
want to be able to cite the core libraries which are used. But I guess it
would clutter things mixing top-level references with say a numpy reference
(at least in the reference.html page that we want to generate) ... ? Also,
to what depth do you recurse? mne-python uses pysurfer which uses mayavi
which uses numpy etc etc. See what I mean?

yarikoptic, 2015-09-05:
re report.py  look on that gist  you posted with error - I   gave explanation there

yarikoptic, 2015-09-05:
changing  habits:  we can  support anything   non-ambiguously providing reference. eg  as long as text form  has identifiable  doi -  we could use that.  lazy me prefers doi  anyway f or specifying  citations for duecredit

Eric89GXL, 2015-12-21:
@jasmainak just to catch up on this -- do we only need to add DOIs to our refs to get this working? I'd prefer to keep the plain text ones in the code if we can.

jasmainak, 2015-12-21:
I think we can close this issue. @agramfort ?

closed 2015-12-21 by agramfort