mne-tools/mne-python #2442: QT backend broken?

kambysese, 2015-09-01:
I am on the development and latest master version, running matplotlib 1.4.3 (Anaconda python env; 64bit Linuxmint 17 OS)  I noticed that `raw.plot()` stopped working a few days ago. Tried running plot_read_and_write_raw_data and plot_morph_data on the example dataset using Qt4agg + PyQt4 and Pyside bindings. Both scripts ran without error but failed to generate figures. Switching to TkAgg works but I lose all matplotlib interactivity (i.e., figures pop open and close).  Any one have any ideas?

agramfort, 2015-09-01:
can you track the commit that broke something?

did you change anything to your python config?

kambysese, 2015-09-02:
@agramfort I am not how to provide what you're asking for. My best guess is <i>Merge pull request #2424 from Eric89GXL/h5io</i> based on the timeline of events, and to a lesser extent on the `conda install...` edits in this commit by @Eric89GXL, but that is likely an installation related change. Other than that I haven't done anything apart from routine updates to conda-python modules {numpy, scipy, and matlaplotlib}.

agramfort, 2015-09-02:
> @agramfort I am not how to provide what you're asking for. My best guess is Merge pull request #2424 from Eric89GXL/h5io based on the timeline of events, and to a lesser extent on the conda install... edits in this commit by @Eric89GXL, but that is likely an installation related change.

I doubt it's related. The conda edits affect travis.

> Other than that I haven't done anything apart from routine updates to conda-python modules {numpy, scipy, and matlaplotlib}.

it can be a conda update of matplotlib. What version are you running?

kambysese, 2015-09-02:
@agramfort most likely a matplotlib issue. I rolled back mne-python back to b143c6d to try and identify a 'bad' branch for bisection but was unable to revert the bug. Next I tried rolling back matplotlib back to 1.3.1 from current 1.4.3; that just led to a [known](https://github.com/ContinuumIO/anaconda-issues/issues/411) shared library problem that involved downgraded pyqt qt numpy et al dependencies.
Now I am back to square one with matplotlib 1.4.3 (Qt4agg+PyQt backend & bindings) which I prefer over a library issue.

kambysese, 2015-09-02:
I think I am hitting [this](https://github.com/matplotlib/matplotlib/issues/3505/)

agramfort, 2015-09-03:
hum...

kambysese, 2015-09-09:
I'll go ahead and close this since it appears to be fixed on my end.

closed 2015-09-09 by kambysese