mne-tools/mne-python #2446: Deprection of Enforce ints for events?

kingjr, 2015-09-02:
This relates to https://github.com/mne-tools/mne-python/pull/2415
@Eric89GXL @agramfort I think we should add a deprecation warning to indicate that from now on mne won't accept non integers in the `events` array, and, in the meantime, try to convert it. I personally have several scripts that now crash because of a float array passed to `EpochsArray()`. I'm of course fine with correcting my scripts, but I suppose that I won't be the only one facing this issue.

dengemann, 2015-09-02:
This is difficult. There is nothing we deprecated in the strict sense. You normally deprecate a function or a parameter. A deprecation warning would require us to still provide the functionality.
It almost feels like a bug that we accepted float values in the first place.

kingjr, 2015-09-02:
I agree with you conceptually. I just think that in practice users may be annoyed.

Eric89GXL, 2015-09-02:
I also don't think it quite applies as a deprecation, but rather as a bug fix. In other words, floating point values were never meant to be supported, and it was an error on our part not to type-check them previously. In this sense, that they worked at all in previous versions can be considered a matter of unfortunate happenstance. For example, you can't read and write such things to disk (here on `0.9`):
```
>>> import numpy as np
>>> import mne
>>> events = np.zeros((1, 3))
>>> events[:, 2] = 1.5
>>> event_id = dict(foo=1.5)
>>> mne.EpochsArray(np.zeros((1, 1, 2)), mne.create_info(1, 1.), events,
                    event_id=event_id).save('temp-epo.fif')
>>> mne.read_epochs('temp-epo.fif')
...
  File "/home/larsoner/custombuilds/mne-python/mne/event.py", line 185, in _read_events_fif
    mappings = dict((k, int(v)) for v, k in m_)
  File "/home/larsoner/custombuilds/mne-python/mne/event.py", line 185, in <genexpr>
    mappings = dict((k, int(v)) for v, k in m_)
ValueError: invalid literal for int() with base 10: '1.5'
>>> Traceback (most recent call last):
  File "/home/larsoner/Desktop/untitled0.py", line 15, in <module>
    print(mne.read_epochs('temp-epo.fif').events)
  File "<string>", line 2, in read_epochs
```
Beyond such clearly failing cases, perhaps more concerning is that floats might also fail to work properly in other cases in some silent or in unexpected way that doesn't throw an exception. For example, there could be places we make use of `isinstance(..., int)` which won't do what it's supposed to with a `float` input, but will instead proceed into a branch that does not raise an error but instead does something silently bad.

If by "convert" you mean convert to integers on input, unfortunately we have no guarantee that previous behavior without such conversion would be maintained. We can't guarantee that such users didn't have events like 1.0, 1.01, 1.02, etc. that would all convert to the same event using a simple `round`; we can't know if they were doing arithmetic on event values so we can't e.g. convert the `float64` bytes to `int64` (which would otherwise preserve uniqueness), and so forth. Unfortunately a warning saying we are doing some simple conversion probably won't be sufficient to prevent users who were using `floats` from mistakenly proceeding with analyses. Warnings are a lot easier to ignore than errors, especially if a script has a lot of logging output. I know it's annoying to have to convert a bunch of scripts, but it really should be safer to force each user to convert their scripts in a way that is appropriate for their data.

kingjr, 2015-09-02:
Ok, I'm convinced, thanks for this detailed response.

closed 2015-09-02 by kingjr