mne-tools/mne-python #2447: BUG: reading cropped epochs

dengemann, 2015-09-02:
We've been living with some bug or bad design for almost 2 years now, according to git praise.
If we crop epochs and the tmin is getting smaller than the baseleine[0] we cannot read them anymore thanks to this check:
https://github.com/mne-tools/mne-python/blame/master/mne/epochs.py#L238

We should drop this check or ignore it when reading eopchs.

Thoughts @agramfort @Eric89GXL @mainakjas ?

Eric89GXL, 2015-09-02:
I think you're right that we should ignore it when reading epochs, because by then the baseline has already been applied and it's just kept around for record-keeping reasons. So maybe add a `check_baseline=True` kwarg to the `_BaseEpochs` constructor that you set to `False` when reading from disk?

dengemann, 2015-09-02:
Yes, that sounds like a solution.

Eric89GXL, 2015-09-02:
BTW I didn't know you could do the `blame` URL with GitHub, that's pretty sweet :)

agramfort, 2015-09-02:
fine with me

agramfort, 2015-10-19:
is this fixed?

Eric89GXL, 2015-10-19:
This runs fine, please reopen issue if it doesn't capture what you meant / it still fails upon modification @dengemann:
```
import numpy as np
import mne
temp_fname = 'test-epo.fif'
info = mne.create_info(1, 1000.)
data = np.zeros((1, 1, 1000))
events = np.zeros((1, 3), int)
events[:, 2] = 1
epochs = mne.EpochsArray(data, info, events, tmin=-0.2,
                         baseline=(None, 0))
epochs.crop(0, None)
epochs.save(temp_fname)
epochs_read = mne.read_epochs(temp_fname)
```

closed 2015-10-19 by Eric89GXL