mne-tools/mne-python #2448: FIX/ENH: Xdawn does not regularize by default when fitting epochs

dengemann, 2015-09-02:
As a result the signal cov can end up not being positive definite, which required by the tanh.
I would suggest to use mne.compute_coveriance with shurnk if epochs are passed.

dengemann, 2015-09-02:
cc @alexandrebarachant

kingjr, 2015-11-25:
Just encountered the issue, so +1 for the shrunk as default.

kingjr, 2015-11-25:
Related to this, shall we pass a `method` arg to `compute_raw_covariance`, to make it work with the [example](http://martinos.org/mne/dev/auto_examples/preprocessing/plot_xdawn_denoising.html#sphx-glr-auto-examples-preprocessing-plot-xdawn-denoising-py)?

dengemann, 2015-11-25:
For raw we don't have / need the method param...

larsoner, 2018-04-04:
I think the error message is clear enough (about semidef, telling you to use reg) as is the solution (pass reg param) so that the deprecation cycle necessary to fix this is not worth it. Closing, but feel free to reopen if you disagree

closed 2018-04-04 by larsoner