mne-tools/mne-python #2450: ENH/VIZ: better default colors for EOG in epochs butterfly plot

dengemann, 2015-09-03:
They're black as EEG and it can be difficult to understand what's going on in cases where the EOG channels have some problem with offsets.

jona-sassenhagen, 2015-09-03:
How about
[image]

The hard part is actually the topomaps on top, the spatial colors for the channels are 2 lines and the bottom left legend is not that complicated.

dengemann, 2015-09-03:
Ahh --- epochs butterfly --- the epochs viewer, when you hit 'b' to look at single trial butterflies, the EOG and the EEG are both black.

jona-sassenhagen, 2015-09-03:
Oh yes, I was thinking of evokeds ... (I never use any of the interactive functions.) But couldn't you do the same thing still?

dengemann, 2015-09-03:
> Oh yes, I was thinking of evokeds ... (I never use any of the interactive functions.) But couldn't you do the same thing still?

It's a different problem. I think you'll better understand it once you start looking at your data using epochs.plot ;-)

agramfort, 2017-07-22:
@dengemann can we close this?

larsoner, 2019-08-20:
Closing due to non-response

closed 2019-08-20 by larsoner