mne-tools/mne-python #2464: epochs rejected even when reject is None

mainakjas, 2015-09-09:
Found a bug in master (I think) after hours of bug-hunting in my own scripts today. Here is the code to reproduce. Look carefully at the console output when you run this script.
```py
import mne
from mne.datasets import sample

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = mne.io.Raw(raw_fname)

picks = mne.pick_types(raw.info, meg=True, eeg=True, stim=False, eog=True,
                       exclude='bads')

events = mne.event.make_fixed_length_events(raw, id=1, duration=1.0)
event_id, tmin, tmax = 1, -0.2, 0.5

epochs = mne.Epochs(raw, events, event_id, tmin, tmax,
                    picks=picks, baseline=(None, 0), reject=None, flat=None,
                    verbose=False)
print(epochs.events.shape)
epochs.drop_bad_epochs()
print(epochs.events.shape)
```
wdyt @agramfort @dengemann @Eric89GXL ?

Eric89GXL, 2015-09-09:
The number of events decreased by one. I think this is the correct behavior if one of the epochs is dropped, right?

Eric89GXL, 2015-09-09:
`epochs.selection` tells you which indices into the original (non-dropped) events list you'd need to use for the given epochs.

mainakjas, 2015-09-09:
but why is it dropped in the first place when `reject `is `None` and `flat` is `None` ?

Eric89GXL, 2015-09-09:
```
>>> epochs.drop_log
[['NO_DATA'], [], [], ...
```
I guess you'll have to investigate what that means. In this case, I suspect the first event is at t=0 but you have tmin=-0.2, so there would be no data for the range (-0.2, 0).

mainakjas, 2015-09-09:
oh dear oh dear, my bad. Indeed, fixing the first event fixes it. Good tip and sorry for the false alarm

closed 2015-09-09 by mainakjas

Eric89GXL, 2015-09-09:
False alarms are better than misses, your detector is biased appropriately :)