mne-tools/mne-python #2465: FIX?: vague user reports, raw.resample destroys trigger channels

dengemann, 2015-09-10:
and generate massiv rejections of epochs with "TOO_SHORT" flag in the drop log.

cc @Eric89GXL @mainakjas ...
cc @cmoutard

agramfort, 2015-09-10:
cc @wmvanvliet 

it was changed during the sprint AFAIK

wmvanvliet, 2015-09-10:
@dengemann details please?

dengemann, 2015-09-10:
> @dengemann details please?

people around here resample their data and script that used to run cras how as 90% of the epochs get rejected. See above. I did not have the problem so far.

agramfort, 2015-09-11:
can they share a file + gist?

wmvanvliet, 2015-09-12:
The new version of the code is less likely to drop event markers. Depending on how the epochs are cut, TOO_SHORT epochs could maybe arise when a buggy stimulus channel shows an on-off-on-off pattern of the event marker?

I'm sorry without any sort of details, there's no way I can debug this.

jona-sassenhagen, 2016-06-29:
Should probably close this. It's likely it was the thing @wmvanvliet fixed during the sprint.

closed 2016-06-29 by Eric89GXL