mne-tools/mne-python #2467: Function for rename split files

teonlamont, 2015-09-10:
I think everyone (at least me) would love us if there was a function allowed for an easy way to rename split files e.g. raw, epochs. I think it should be rather straightforward: we would just need to just get find a list of all the files, and figure out where this info is written, probably at some specific tag location, and overwrite it there. We wouldn't need to load data or anything.

Currently the solutions are 1. do nothing and deal with the file now, or 2. redo the saving all over again included writing data.

Eric89GXL, 2015-09-11:
I thought if you renamed both files in the same way but maintained the `-1.fif` relationship they would still be found, at least in the raw case. Is that not true?

teonlamont, 2015-09-11:
no, there's a reference to the original file name in `inst.info['filename']`, which poses a reference problem for `-1.fif` files

agramfort, 2015-09-11:
it would mean being able to update the info on disk without loading the
data.

can't you live with loading and saving again?

teonlamont, 2015-09-12:
ah, i thought it was possible to rewrite parts of the info of a file. i'm fine if that's not currently possible

closed 2015-09-12 by teonlamont