mne-tools/mne-python #2469: plot_topomap doesnt work if colorbar is given

ImmanuelSamuel, 2015-09-12:
Exactly as title says. It works if colorbar=None

ImmanuelSamuel, 2015-09-12:
[image]

ImmanuelSamuel, 2015-09-12:
[image]

ImmanuelSamuel, 2015-09-12:
Yes I know the above can be achieved by doing
```
fig, axes = plt.subplots(nrows=1, ncols=6, figsize=(4,2))
evoked.plot_topomap([0,1,2,3,4,5],vmin=-200,vmax=200,axes=axes,colorbar=None)
```
But I need to be able to plot different in evoked datasets in the same figure but in different subplots. That is the reason why I wanted this enhancement.

ImmanuelSamuel, 2015-09-13:
As you can see in Ipython notebook. The function plots it twice.

agramfort, 2015-09-13:
@jaeilepp can you have a look?

jaeilepp, 2015-09-14:
> But I need to be able to plot different evoked datasets in the same figure but in different subplots. That is the reason why I wanted this enhancement.
For example the following doesnt work

> fig, axes = plt.subplots(nrows=2, ncols=6, figsize=(4,2))
evoked.plot_topomap([0,1,2,3,4,5],axes=axes[1,:],colorbar=None)
evoked1.plot_topomap([0,1,2,3,4,5],axes=axes[0,:],colorbar=None)

You can use ``mne.viz.topomap.plot_evoked_topomap`` to achieve this.
``fig, axes = plt.subplots(nrows=2, ncols=6, figsize=(4,2))
mne.viz.topomap.plot_evoked_topomap(evoked, show=False,  axes=axes[0])
mne.viz.topomap.plot_evoked_topomap(evoked1, show=True,  axes=axes[1]))``

ImmanuelSamuel, 2015-09-14:
I dont want to plot all evoked time points

Even if I do the following it doesnt work
[image]

ImmanuelSamuel, 2015-09-14:
This code 
```
import mne
import numpy as np
from matplotlib import pyplot as plt
%matplotlib inline

# Channel positions
ch_names = ['A1', 'A2', 'A3', 'A4', 'A5', 'A6', 'A7', 'A8', 'A9', 'A10', 'A11', 'A12', 'A13', 'A14', 'A15', 'A16', 'A17', 'A18', 'A19', 'A20', 'A21', 'A22', 'A23', 'A24', 'A25', 'A26', 'A27', 'A28', 'A29', 'A30', 'A31', 'A32', 'B1', 'B2', 'B3', 'B4', 'B5', 'B6', 'B7', 'B8', 'B9', 'B10', 'B11', 'B12', 'B13', 'B14', 'B15', 'B16', 'B17', 'B18', 'B19', 'B20', 'B21', 'B22', 'B23', 'B24', 'B25', 'B26', 'B27', 'B28', 'B29', 'B30', 'B31', 'B32', 'C1', 'C2', 'C3', 'C4', 'C5', 'C6', 'C7', 'C8', 'C9', 'C10', 'C11', 'C12', 'C13', 'C14', 'C15', 'C16', 'C17', 'C18', 'C19', 'C20', 'C21', 'C22', 'C23', 'C24', 'C25', 'C26', 'C27', 'C28', 'C29', 'C30', 'C31', 'C32', 'D1', 'D2', 'D3', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'D10', 'D11', 'D12', 'D13', 'D14', 'D15', 'D16', 'D17', 'D18', 'D19', 'D20', 'D21', 'D22', 'D23', 'D24', 'D25', 'D26', 'D27', 'D28', 'D29', 'D30', 'D31', 'D32']
montage = mne.channels.read_montage('biosemi128')
info = mne.create_info(ch_names, 100, 'eeg', montage)

# EEG data
temp = [-0.00054827592108, 0.031668073094, 0.0704223162929, 0.0865540229014, 0.085150043403, 0.0457380793299, 0.0509495732895, 0.0987022550764, 0.0801886633895, 0.04976969651, 0.00614093831533, -0.0373887332355, -0.0197260623702, -0.000444422313751, 0.0248963507643, 0.0589171499815, 0.0717817674042, 0.0787944587508, 0.130753138626, 0.124899705387, 0.0975269669664, 0.0419575393272, -0.00793216362559, -0.0151265531253, -0.0348081764706, 0.0215552729428, 0.0249141780785, 0.0655540808354, 0.108124467678, 0.124820721722, 0.146384148659, 0.140718598747, 0.0199164974076, 0.0571570915027, 0.0876578355539, 0.0923016839654, 0.133136594185, 0.152798531971, 0.122602552914, 0.0714821347463, 0.0537612111322, 0.0469385417282, 0.0740553145655, 0.0915974711193, 0.0743604227044, 0.0110220331254, 0.0234080078001, 0.0325478142343, 0.0477724066775, 0.047792830146, 0.0666905921563, 0.0189050684572, 0.00897804762235, 0.0227890832216, 0.0197423673457, -0.00454527858583, -0.00615177981404, -0.0201675075456, -0.0920040702005, -0.0464443873328, -0.0414552367577, -0.0531930278853, -0.0307912429832, -0.0412551750461, -0.0123007168696, -0.0112167955503, -0.0542948025463, -0.0857190408132, -0.0771482839, -0.113035664, -0.124491501499, -0.122871051658, -0.10429939481, -0.0749259731853, -0.0494703403965, -0.0760484386394, -0.0866986470174, -0.108261995558, -0.103236351804, -0.0826894600891, -0.114552515316, -0.0938510344287, -0.0872763177219, -0.0658120109097, -0.109361027459, -0.0545709142688, -0.0465159668016, -0.0374219872666, -0.0657419626256, -0.0692102438198, -0.0685479219719, -0.0770634783383, -0.0872426401117, -0.0666948763193, -0.0573920809363, -0.0800195158044, -0.0249282154474, -0.0257314683258, -0.0342883027284, -0.0602967350127, -0.0379301562186, -0.0422883555211, -0.0357673248702, -0.00463023885605, -0.0154207203639, -0.0236424654065, -0.00446259655858, 0.00180786601789, -0.000400985259411, 0.000962388449819, 0.0113404811841, 0.0430510059066, 0.0267019564535, -0.00120974681118, 0.00317540453295, 0.00662862089829, -0.0460368212177, -0.0417816042596, 0.00620115910578, -0.0218318617073, -0.0349000304808, -0.0231747760115, 0.0320176071901, 0.00236553571098, 0.0207500486866, 0.0235998503673, -0.00664424438941, -0.0175665711621]

l = mne.channels.make_eeg_layout(info)

# Make topomap plots
fig, axes = plt.subplots(nrows=1, ncols=3, figsize=(7,3))

ax = axes[0]
mne.viz.topomap.plot_topomap(temp,l.pos,axis=ax)

ax = axes[1]
mne.viz.topomap.plot_topomap(temp,l.pos,axis=ax)

ax = axes[2]
mne.viz.topomap.plot_topomap(temp,l.pos,axis=ax)
```
Still doesn't work
[image]

ImmanuelSamuel, 2015-09-14:
The problem of everything plotting twice it still there. 
Also the right most subplot is the colorbar which is not correct because it omits a time instance.

jaeilepp, 2015-09-15:
>The problem of everything plotting twice it still there.

You can prevent the plot from showing prematurely by giving ``show=False`` to first two calls of ``mne.viz.topomap.plot_topomap``.
>Also the right most subplot is the colorbar which is not correct because it omits a time instance.

You can either give more space for the colorbar by making an extra column for the figure or giving ``colorbar=False`` to ``plot_evoked_topomap``.

ImmanuelSamuel, 2015-09-16:
Thanks for the show=False tip. Solves most of the problems

ImmanuelSamuel, 2015-10-23:
How to show a colorbar for ```mne.viz.plot_topomap```?
How to get the colorbar from the plot returned by ```plot_evoked_topomap``` or ```evoked.plot_topomap``` and plot it a different axis?

agramfort, 2015-10-25:
please have a look at the code. It needs a bit of work. If you struggle ask
again. And if you could share your progress with a PR it would be great.

jona-sassenhagen, 2015-11-09:
@ImmanuelSamuel : Cause I'm lazy, to suppress double showing of the last element in `%matplotlib inline` notebooks, I usually do this:

```_ = plot_something(...```

(The notebook will return the last returned element by default, and if it's a figure, it will show it. If you feed it to an argument, you avoid this double plot.)

agramfort, 2015-11-09:
or just end the line with ;

jona-sassenhagen, 2015-11-09:
Oh, I didn't know that! That's even lazier :)

Eric89GXL, 2015-12-21:
@ImmanuelSamuel I suspect that both of those colorbar issues can be solved by looking at `matplotlib` Figure properties and the like. Feel free to reopen if you think these use cases should be part of `mne-python`

closed 2015-12-21 by Eric89GXL

ImmanuelSamuel, 2015-12-22:
Yeah ok thanks for the tips. Right now I am just plotting some random stuff in an axes and getting the colorbar handle. I hide that axes after that.