mne-tools/mne-python #2470: RuntimeError: info channel name inconsistency detected, please notify mne-python developers

devadidev, 2015-09-14:
Hi, I'm new to MNE and Python and I've been trying to run the example plot_decoding_csp_eeg.py on my Ubuntu 14.04, Anaconda with conda 3.17.0 and MNE:  0.10.dev0    <pip>.  I'm getting the following error and wondering if Im missing something trivial! Thanks

(mnemeeg)➜  programs :  python plot_decoding_csp_eeg.py 

===========================================================================
Motor imagery decoding from EEG data using the Common Spatial Pattern (CSP)
===========================================================================

Decoding of motor imagery applied to EEG data decomposed using CSP.
Here the classifier is applied to features extracted on CSP filtered signals.

See http://en.wikipedia.org/wiki/Common_spatial_pattern and [1]

The EEGBCI dataset is documented in [2]
The data set is available at PhysioNet [3]

[1] Zoltan J. Koles. The quantitative extraction and topographic mapping
    of the abnormal components in the clinical EEG. Electroencephalography
    and Clinical Neurophysiology, 79(6):440--447, December 1991.

[2] Schalk, G., McFarland, D.J., Hinterberger, T., Birbaumer, N.,
    Wolpaw, J.R. (2004) BCI2000: A General-Purpose Brain-Computer Interface
    (BCI) System. IEEE TBME 51(6):1034-1043

[3] Goldberger AL, Amaral LAN, Glass L, Hausdorff JM, Ivanov PCh, Mark RG,
    Mietus JE, Moody GB, Peng C-K, Stanley HE. (2000) PhysioBank,
    PhysioToolkit, and PhysioNet: Components of a New Research Resource for
    Complex Physiologic Signals. Circulation 101(23):e215-e220

Extracting edf Parameters from /home/devanka/src/examples/MNE-eegbci-data/physiobank/database/eegmmidb/S001/S001R06.edf...
Setting channel info structure...
Creating Raw.info structure...
Reading 0 ... 19999  =      0.000 ...   124.994 secs...
[done]
Ready.
Extracting edf Parameters from /home/devanka/src/examples/MNE-eegbci-data/physiobank/database/eegmmidb/S001/S001R10.edf...
Setting channel info structure...
Creating Raw.info structure...
Reading 0 ... 19999  =      0.000 ...   124.994 secs...
[done]
Ready.
Extracting edf Parameters from /home/devanka/src/examples/MNE-eegbci-data/physiobank/database/eegmmidb/S001/S001R14.edf...
Setting channel info structure...
Creating Raw.info structure...
Reading 0 ... 19999  =      0.000 ...   124.994 secs...
[done]
Ready.
Traceback (most recent call last):
  File "plot_decoding_csp_eeg.py", line 62, in <module>
    raw.filter(7., 30., method='iir')
  File "<string>", line 2, in filter
  File "/home/devanka/projects/mneruns/programs/src/mne/mne/utils.py", line 540, in verbose
    return function(*args, **kwargs)
  File "/home/devanka/projects/mneruns/programs/src/mne/mne/io/base.py", line 804, in filter
    picks = pick_types(self.info, exclude=[], **pick_parameters)
  File "/home/devanka/projects/mneruns/programs/src/mne/mne/io/pick.py", line 205, in pick_types
    info._check_consistency()
  File "/home/devanka/projects/mneruns/programs/src/mne/mne/io/meas_info.py", line 219, in _check_consistency
    raise RuntimeError('info channel name inconsistency detected, '
RuntimeError: info channel name inconsistency detected, please notify mne-python developers
(mnemeeg)➜  programs  python plot_decoding_csp_eeg.py

===========================================================================
Motor imagery decoding from EEG data using the Common Spatial Pattern (CSP)
===========================================================================

Decoding of motor imagery applied to EEG data decomposed using CSP.
Here the classifier is applied to features extracted on CSP filtered signals.

See http://en.wikipedia.org/wiki/Common_spatial_pattern and [1]

The EEGBCI dataset is documented in [2]
The data set is available at PhysioNet [3]

[1] Zoltan J. Koles. The quantitative extraction and topographic mapping
    of the abnormal components in the clinical EEG. Electroencephalography
    and Clinical Neurophysiology, 79(6):440--447, December 1991.

[2] Schalk, G., McFarland, D.J., Hinterberger, T., Birbaumer, N.,
    Wolpaw, J.R. (2004) BCI2000: A General-Purpose Brain-Computer Interface
    (BCI) System. IEEE TBME 51(6):1034-1043

[3] Goldberger AL, Amaral LAN, Glass L, Hausdorff JM, Ivanov PCh, Mark RG,
    Mietus JE, Moody GB, Peng C-K, Stanley HE. (2000) PhysioBank,
    PhysioToolkit, and PhysioNet: Components of a New Research Resource for
    Complex Physiologic Signals. Circulation 101(23):e215-e220

Extracting edf Parameters from /home/devanka/src/examples/MNE-eegbci-data/physiobank/database/eegmmidb/S001/S001R06.edf...
Setting channel info structure...
Creating Raw.info structure...
Reading 0 ... 19999  =      0.000 ...   124.994 secs...
[done]
Ready.
Extracting edf Parameters from /home/devanka/src/examples/MNE-eegbci-data/physiobank/database/eegmmidb/S001/S001R10.edf...
Setting channel info structure...
Creating Raw.info structure...
Reading 0 ... 19999  =      0.000 ...   124.994 secs...
[done]
Ready.
Extracting edf Parameters from /home/devanka/src/examples/MNE-eegbci-data/physiobank/database/eegmmidb/S001/S001R14.edf...
Setting channel info structure...
Creating Raw.info structure...
Reading 0 ... 19999  =      0.000 ...   124.994 secs...
[done]
Ready.
Traceback (most recent call last):
  File "plot_decoding_csp_eeg.py", line 62, in <module>
    raw.filter(7., 30., method='iir')
  File "<string>", line 2, in filter
  File "/home/devanka/projects/mneruns/programs/src/mne/mne/utils.py", line 540, in verbose
    return function(*args, **kwargs)
  File "/home/devanka/projects/mneruns/programs/src/mne/mne/io/base.py", line 804, in filter
    picks = pick_types(self.info, exclude=[], **pick_parameters)
  File "/home/devanka/projects/mneruns/programs/src/mne/mne/io/pick.py", line 205, in pick_types
    info._check_consistency()
  File "/home/devanka/projects/mneruns/programs/src/mne/mne/io/meas_info.py", line 219, in _check_consistency
    raise RuntimeError('info channel name inconsistency detected, '
RuntimeError: info channel name inconsistency detected, please notify mne-python developers

kingjr, 2015-09-14:
Have you tried with ipython instead of python? Check your mne version inside ipython:
```import mne
print mne.__version__```

mainakjas, 2015-09-14:
looks related to changes in PR https://github.com/mne-tools/mne-python/pull/2452

devadidev, 2015-09-14:
@kingjr, the version returned is as I quoted in the original, I tried with iPython and got a different (qt related?) error. I'll explore this further and update here if I'm able to configure and run it successfully also @mainakjas thanks for linking the issue, I'll go through that update here if it's related to that.

iPython Error:
   computing connectivity for epoch 58
    computing connectivity for epoch 59
    computing connectivity for epoch 60
[Connectivity computation done]
Computing PSI from estimated Coherency
[PSI Estimation Done]
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
/home/devanka/projects/mneruns/programs/plot_mne_inverse_psi_visual.py in <module>()
    110                      time_label='Phase Slope Index (PSI)',
    111                      subjects_dir=subjects_dir,
--> 112                      clim=dict(kind='percent', pos_lims=(95, 97.5, 100)))
    113 brain.show_view('medial')
    114 brain.add_label(fname_label, color='green', alpha=0.7)

/home/devanka/projects/mneruns/programs/src/mne/mne/source_estimate.pyc in plot(self, subject, surface, hemi, colormap, time_label, smoothing_steps, transparent, alpha, time_viewer, config_opts, subjects_dir, figure, views, colorbar, clim)
   1362                                       subjects_dir=subjects_dir, figure=figure,
   1363                                       views=views, colorbar=colorbar,
-> 1364                                       clim=clim)
   1365         return brain
   1366 

/home/devanka/projects/mneruns/programs/src/mne/mne/viz/_3d.pyc in plot_source_estimates(stc, subject, surface, hemi, colormap, time_label, smoothing_steps, transparent, alpha, time_viewer, config_opts, subjects_dir, figure, views, colorbar, clim)
    519         A instance of surfer.viz.Brain from PySurfer.
    520     """
--> 521     from surfer import Brain, TimeViewer
    522     config_opts = _handle_default('config_opts', config_opts)
    523 

/home/devanka/anaconda/envs/mnemeeg/lib/python2.7/site-packages/surfer/__init__.py in <module>()
----> 1 from .viz import Brain, TimeViewer  # noqa
      2 from .utils import Surface, verbose, set_log_level, set_log_file  # noqa
      3 from .io import project_volume_data  # noqa
      4 
      5 __version__ = "0.6"

/home/devanka/anaconda/envs/mnemeeg/lib/python2.7/site-packages/surfer/viz.py in <module>()
     12 import nibabel as nib
     13 
---> 14 from mayavi import mlab
     15 from mayavi.tools.mlab_scene_model import MlabSceneModel
     16 from mayavi.core import lut_manager

/home/devanka/anaconda/envs/mnemeeg/lib/python2.7/site-packages/mayavi/mlab.py in <module>()
     25 
     26 # Mayavi imports
---> 27 from mayavi.tools.camera import view, roll, yaw, pitch, move
     28 from mayavi.tools.figure import figure, clf, gcf, savefig, \
     29     draw, sync_camera, close, screenshot

/home/devanka/anaconda/envs/mnemeeg/lib/python2.7/site-packages/mayavi/tools/camera.py in <module>()
     23 # We can't use gcf, as it creates a circular import in camera management
     24 # routines.
---> 25 from engine_manager import get_engine
     26 
     27 

/home/devanka/anaconda/envs/mnemeeg/lib/python2.7/site-packages/mayavi/tools/engine_manager.py in <module>()
     10 
     11 # Local imports
---> 12 from mayavi.preferences.api import preference_manager
     13 from mayavi.core.registry import registry
     14 from mayavi.core.engine import Engine

/home/devanka/anaconda/envs/mnemeeg/lib/python2.7/site-packages/mayavi/preferences/api.py in <module>()
      2 
      3 # The global PreferenceManager instance
[... 77 lines cut ...]
     52 
     53 

/home/devanka/anaconda/envs/mnemeeg/lib/python2.7/site-packages/traitsui/qt4/__init__.py in <module>()
     16 # import pyface.qt before anything else is done so the sipapi
     17 # can be set correctly if needed
---> 18 import pyface.qt
     19 
     20 #----------------------------------------------------------------------------

/home/devanka/anaconda/envs/mnemeeg/lib/python2.7/site-packages/pyface/qt/__init__.py in <module>()
     26     except ImportError:
     27         try:
---> 28             prepare_pyqt4()
     29             import PyQt4
     30             qt_api = 'pyqt'

/home/devanka/anaconda/envs/mnemeeg/lib/python2.7/site-packages/pyface/qt/__init__.py in prepare_pyqt4()
     15     # Set PySide compatible APIs.
     16     import sip
---> 17     sip.setapi('QString', 2)
     18     sip.setapi('QVariant', 2)
     19 

ValueError: API 'QString' has already been set to version 1

mainakjas, 2015-09-14:
@devadidev for the error, have a look here: https://pysurfer.github.io/install.html. It should fix it

devadidev, 2015-09-14:
@mainakjas, thanks that fixed the iPython qt error but back to the RunTime one:

/home/devanka/projects/mneruns/programs/src/mne/mne/io/meas_info.pyc in _check_consistency(self)
    217                 ch_1 != ch_2 for ch_1, ch_2 in zip(self['ch_names'], chs)) or \
    218                 self['nchan'] != len(chs):
--> 219             raise RuntimeError('info channel name inconsistency detected, '
    220                                'please notify mne-python developers')
    221         # make sure we have the proper datatypes

RuntimeError: info channel name inconsistency detected, please notify mne-python developers

agramfort, 2015-09-14:
plot_decoding_csp_eeg.py runs fine for me in master. Are you using master?

what version of python are you using?

mainakjas, 2015-09-14:
where did you get the example from? Did you download it from the website or github?

mainakjas, 2015-09-14:
it runs fine for me too

devadidev, 2015-09-14:
Ah ic now, I downloaded from the website (http://martinos.org/mne/stable/auto_examples/decoding/plot_decoding_csp_eeg.html) and there are some diff with GH version of the code, which runs fine for me in master now.

(mnemeeg)➜  programs  diff plot_decoding_csp_eeg_old.py plot_decoding_csp_eeg_github.py 
5d4
< 
8d6
< 
10d7
< 
13d9
< 
17d12
< 
21d15
< 
34,36c28,30
< from mne import Epochs, pick_types
< from mne.io import concatenate_raws
< from mne.io.edf import read_raw_edf
---
> from mne import Epochs, pick_types, find_events
> from mne.channels import read_layout
> from mne.io import concatenate_raws, read_raw_edf
38d31
< from mne.event import find_events
40d32
< from mne.channels import read_layout
58,59c50,51
< # strip channel names
< raw.info['ch_names'] = [chn.strip('.') for chn in raw.info['ch_names']]
---
> # strip channel names of "." characters
> raw.rename_channels(lambda x: x.strip('.'))

closed 2015-09-14 by mainakjas

agramfort, 2015-09-14:
ok please close then.