mne-tools/mne-python #2471: Add info about quadrature rules in the forward solution file

hsonntag, 2015-09-15:
During the computation of a forward solution for MEG, the Integral of the magnetic flux on each sensor area is usually approximated using quadrature rules. MNE offers different orders of quadrature rules which provide different accuracies. However, no information about the quadrature rules is stored in the forward solution files.
I suggest to add this information for example via the number of quadrature nodes/weights for each sensor. 
Further information about the coordinates of the nodes and the values of the weights can be found in the coil_def.dat if the order of the quadrature rule is known and thus it is unnecessary to add this info too, right? On the other hand it might be convenient to have the coordinates and weights right away.

agramfort, 2015-09-15:
did you look in _make_forward.py if you can find this information?

Eric89GXL, 2015-09-15:
I think it's unnecessary to add the information to the file, but it would
be beneficial to add it to the documentation somewhere. The Maxwell
filtering algorithm uses a different set of integration points than the
standard coil_def, and it would be great to have these things all
documented in one place. @wronk did you end up adding documentation about
these issues in one of your PRs? I am AFK so can't check easily.

wronk, 2015-09-15:
Ya, check `/mne/data/coil_def_Elekta.dat`. I had to use different integration points to get SSS to match up with Elekta's version. As far as I could tell when working on this, Matti just divided up the MEG coil sensor area evenly into a grid of integration points and also assigned even weights. When SSS was being formulated, they used a flux approximation formula from Abramowitz and Stegun (the formula number is listed in the .dat file). That book doesn't have much in the way of explanation though.

hsonntag, 2015-09-16:
The accuracy is an argument for _prep_channels defined in _make_forward.py and in make_forward_solution this argument is not assigned and the default value of accurate=True is used.
Since only the most accurate of the available orders is used right now, it might be really unnecessary to provide this info yet.
I did not know that there is a standalone python implementation for the forward computation until yesterday but this does not call any C code right?
Only in the C code the accuracy is an argument and it seemed to be useful to preserve it.

agramfort, 2015-09-16:
feel free to close if you feel like it.

yes the forward code is pure python

closed 2015-09-16 by hsonntag

Eric89GXL, 2015-09-16:
It would be easy to add the argument if it would be useful to people. We didn't add it because it made little difference in terms of computation time in Python, so we stuck with the high-accuracy solutions at all times.