mne-tools/mne-python #2472: Bug in toggling the slider

teonlamont, 2015-09-15:
I just noticed that the toggling does not work for bem sliders properly. There are two toggle buttons that appear instead of one and they are doing slightly different things: the `bem` label toggles the table of contents reference but the `MRI` one toggles the sliders. shouldn't this be just one button? 
cc: @mainakjas 

code snippet below:
```python
import mne
from mne import report
rep = report.Report()
path = op.join(mne.datasets.sample.data_path(), 'subjects')
rep.add_bem_to_section('fsaverage', decim=100, subjects_dir=path)
rep.save('~/Desktop/test.html', overwrite=True)
```

mainakjas, 2015-09-15:
yeah, I guess it should all be `MRI`

teonlamont, 2015-09-15:
that just created to buttons with `MRI` as their labels ;)

mainakjas, 2015-09-15:
@teonlamont I am closing this because I realized that `add_bem_to_section` is a "custom" section and "custom" sections are delineated from the "default / auto" sections. This was a deliberate design choice. Feel free to reopen if you disagree or open a PR if you think there can be an exception for BEMs.

closed 2015-09-15 by mainakjas

teonlamont, 2015-09-15:
doesn't this make the add_bem_to_section a bit clunky since we aren't able
to make all parts of the bem available to the custom class. it doesn't seem
like good design to have two tabs dedicated to one section

-teon

mainakjas, 2015-09-15:
With `add_bem_to_section`, the user can add BEMs for as many subjects as they want which is a slightly different use case than having one BEM per report. You can circumvent this problem by using `add_bem_to_section` for both the subjects.