mne-tools/mne-python #2476: Load data from brainstorm/fieldtrip

vborghe, 2015-09-17:
Hello!

We really like the tools provided by mne-python and would love to use them for our analyses. The problem is that our data have already been preprocessed (different stages) with brainstorm/fieldtrip. We haven't been able to find an easy way to import them into an mne-python compatible structure. We turned to the [fieldtrip website](http://www.fieldtriptoolbox.org/development/integrate_with_mne), where it looks like few problems have been solved:

* pre-processed (not epoched) data can be saved as .fif and then loaded in python 
matlab:
    _fiff_file  = 'ctf_raw.fif'_
    _fieldtrip2fiff(fiff_file, data)_
python:
    _from mne.io import Raw_
    _raw = Raw('ctf_raw.fif')_

* evoked responses can be saved as .fif and then loaded in python 
matlab:
   _fiff_file = 'ctf-ave.fif'_
    _fieldtrip2fiff(fiff_file, avg)_
python:
    _from mne.fiff import read_evoked_
    _evoked = read_evoked('ctf-ave.fif')_

However, for data that have been preprocessed and epoched there is currently no suggested procedure (the code is under development - and it has been like that for a while...).

Is there (or will there be) any attempt to facilitate loading of data preprocessed elsewhere into mne-python? Compatibility would increase users :wink: 

If not, would you be willing to give us some help?

We are of course available to share data and provide more details on the fieldtrip structure.

Thank you very much

@vborghe
@grmanxchange
@pinheirochagas

dengemann, 2015-09-17:
I agree it would be good to have an epochs io in Fieldtrip. There is already a dedicated IO function for dealing with epochs in the mne-matlab code, I thought Fieldtrip adopted it already.

For now why don't you just export from matlab the essential data in .mat file and construct the data containers you need. Should be doable in a few lines of code:

http://martinos.org/mne/dev/auto_tutorials/plot_creating_data_structures.html

Before writing a Brainstorm IO it would be good to know how consistent the file format has been so far and which are the planned changes, etc.

Anyway: `scipy.io.loadmat` + mne.EpochsArray are your friends.

mainakjas, 2015-09-17:
@vborghe do you want to share more details regarding what type of preprocessing steps you have already accomplished elsewhere? I'm curious :) Maybe there's an easy solution, or maybe not ...

vborghe, 2015-09-17:
@dengemann Just to clarify: from Brainstorm to Fieldtrip we are (quite) covered. It's already the "standard" to do stats in Fieldtrip after preprocessing in Brainstorm. Also, we're already familiar with loading .mat in python (God save scipy.io!), but we will likely need help on how to (efficiently) build the mne-python structure.

@mainakjas Basically our data are epochs ready for fun things (e.g. decoding). Our (@grmanxchange, @pinheirochagas) pipelines are slightly different, but let's say removal of artifacts, filtering, baseline correction, and downsampling are done.

dengemann, 2015-09-17:
@vborghe :) --> http://martinos.org/mne/dev/auto_tutorials/plot_creating_data_structures.html

mainakjas, 2015-09-17:
I agree with @dengemann that's the way to go @vborghe . Let us know if you need help along the way :)

kingjr, 2015-09-17:
If that helps, here's an example to load mat data save as a fieldtrip structure into mne:

```

def load_mat(fname, var_name='data'):
    from mne import create_info
    from mne.epochs import EpochsArray
    import scipy.io as sio
    import numpy as np
    # load Matlab/Fieldtrip data
    mat = sio.loadmat(fname, squeeze_me=True, struct_as_record=False)
    ft_data = mat[var_name]

    # convert to mne
    n_trial = len(ft_data.trial)
    n_chans, n_time = ft_data.trial[0].shape
    data = np.zeros((n_trial, n_chans, n_time))
    for trial in range(n_trial):
        data[trial, :, :] = ft_data.trial[trial]

    sfreq = float(ft_data.fsample)
    time = ft_data.time[0]

    # only keep mag channels
    # XXX /!\ THIS IS A NON STANDARD STEP ------------------------------
    coi = range(306)
    data = data[:, coi, :]
    chan_names = [l.encode('ascii') for l in ft_data.label[coi]]
    chan_types = ft_data.label[coi]
    chan_types[0:305:3] = 'grad'
    chan_types[1:305:3] = 'grad'
    chan_types[2:306:3] = 'mag'
    # ------------------------------------------------------------------
    info = create_info(chan_names, sfreq, chan_types)
    events = np.zeros((n_trial, 3), int)
    epochs = EpochsArray(data, info, events=events,
                         tmin=np.min(time), verbose=False)
    return epochs, ft_data.trialinfo
```

But yes, it'd be great to make a function that properly reads fieltrip's structure and document it :)

vborghe, 2015-09-17:
@kingjr  that does help and (at a very very quick try) it works. I had to explicitly assign the sfreq [I did brainstorm-to-fieltrip-to-mnepython - yes, there is something wrong with me] but I think @grmanxchange should be fine with this. @pinheirochagas is in my same situation [I'm contagious]

We will try it out _for real_ and keep you posted.
Thanks very much to everyone!

mainakjas, 2015-09-17:
at least you are proceeding in the right direction. GUI -> scripting -> python :)

vborghe, 2015-09-18:
Once adapted few things (due to the fact that mine data and @pinheirochagas already lost some info transitioning from brainstorm to fieldtrip), everything seems to be working OK. [Here](https://gist.github.com/vborghe/76ff9c3deffdeba3c3b0) a (very) simple and fast application which should work for @pinheirochagas and @grmanxchange as well. We will start playing with more things and see whether any problem arise in other settings.

Thanks very much @kingjr: you don't want to see how decoding looks when done step by step outside mne :flushed: 

Should you decide to explore more these compatibility-related issues, our data, use-cases, and doubts are at your disposal :wink: 

[I'll update Lou, she wrote on the mailing list with this same issue]

kingjr, 2015-09-18:
@vborghe You can let contact me directly if you encounter difficulties with decoding. I'm always keen on robust it is in other labs

agramfort, 2015-09-19:
thanks @vborghe <https://github.com/vborghe>  for sharing !

kingjr, 2016-03-25:
@vborghe How about transforming your code to a tutorial to complete http://martinos.org/mne/dev/manual/index.html #migrating?

vborghe, 2016-03-26:
@kingjr if you think it would be worth it, I'd love to help. I got to many things on my plate this weeks, but soon enough I should be able do write something similar to migrating-to-EEGLAB

kingjr, 2016-03-26:
That'd be great yes :)

anaevf, 2018-01-19:
Hey all,

I know this an old thread but I am having an issue when trying to implement the solution here proposed to import epoched data to mne, I get the error:

`load_mat(fname, var_name=var_name)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 26, in load_mat
  File "<string>", line 2, in __init__
  File "C:\Users\Ana\Anaconda2\lib\site-packages\mne\utils.py", line 727, in verbose
    return function(*args, **kwargs)
  File "C:\Users\Ana\Anaconda2\lib\site-packages\mne\epochs.py", line 1981, in __init__
    proj=proj, on_missing=on_missing)
  File "C:\Users\Ana\Anaconda2\lib\site-packages\mne\epochs.py", line 273, in __init__
    raise RuntimeError('Event time samples were not unique')
RuntimeError: Event time samples were not unique`

I am new in python (so this will be probably a silly solution) and I am not being able to debug. Any help would be appreciated!

agramfort, 2018-01-19:
what are you trying to do? what is the data?

anaevf, 2018-01-21:
Hi @agramfort,

I am trying to import an already epoched and cleaned (in Fieldtrip) EEG file to the MNE as an epoch object. My final aim is to try the decoding over time method using the (very much appreciated!) MNE tools. The fieldtrip page is not very helpful, however, saying that the fieldtrip2fiff function does not support already epoched data.

I was really glad when I found this issue here and I am trying to implement the proposed solution:

`def load_evk(fname, var_name='data'):
    from mne import create_info
    from mne.epochs import EpochsArray
    import scipy.io as sio
    import numpy as np
    # load Matlab/Fieldtrip data
    mat = sio.loadmat(fname, squeeze_me=True, struct_as_record=False)
    ft_data = mat[var_name]

    # convert to mne
    n_trial = len(ft_data.trial)
    n_chans, n_time = ft_data.trial[0].shape
    data = np.zeros((n_trial, n_chans, n_time))
    for trial in range(n_trial):
        data[trial, :, :] = ft_data.trial[trial]

    sfreq = float(ft_data.fsample)
    time = ft_data.time[0]

    coi = range(n_chans)
    data = data[:, coi, :]
    chan_names = [l.encode('ascii') for l in ft_data.label[coi]]
    chan_types = ft_data.label[coi]
    chan_types[:] = 'eeg'
    info = create_info(chan_names, sfreq, chan_types)
    events = np.zeros((len(ft_data.trial), 3), int)
    epochs = EpochsArray(data, info, events=events,
                         tmin=np.min(time), verbose=False)
    return epochs, ft_data.trialinfo`

But maybe I am too new at python, but once I try to import my data with the code, I get the following error: 

> Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "<string>", line 2, in __init__
  File "C:\Users\Ana\Anaconda2\lib\site-packages\mne\utils.py", line 727, in verbose
    return function(*args, **kwargs)
  File "C:\Users\Ana\Anaconda2\lib\site-packages\mne\epochs.py", line 1981, in __init__
    proj=proj, on_missing=on_missing)
  File "C:\Users\Ana\Anaconda2\lib\site-packages\mne\epochs.py", line 273, in __init__
    raise RuntimeError('Event time samples were not unique')
RuntimeError: Event time samples were not unique

I think I maybe don't fully understand what the events should look like? I am completely new in Python so I think I might be missing something pretty obvious here.

Thank you for your help!

agramfort, 2018-01-21:
you pass :

events = np.zeros((len(ft_data.trial), 3), int)

rather than passing zeros just set events to None.

Catsine, 2018-09-10:
Hi @agramfort, I am afraid my knowledge of programming is not so advanced, but I will try to read and understand. Thanks a lot!

Catsine, 2018-09-11:
Hello again @agramfort,...I am sure this is a silly question, but once I import my data on python and I run the function used also by @anaevf, I get the message that the np.array has not attribute trial (or either time, label, etc). Is there a library I forgot to import or is it a property of the fieldtrip dataset? I have received the data in a mat format, without doing the pre-processing myself, so that I do not know whether I should before modify something in matlab myself. Thank you very much.

agramfort, 2018-09-11:
I don't know. Maybe if you share the file one of our fieldtrip expert can
have a look

Catsine, 2018-09-12:
Thanks @agramfort, but I managed to solve the issue. The file had been pre-processed in EEGLAB and after adapting vborghe's script I could extract the information needed. Thanks

larsoner, 2019-08-20:
We have FT I/O now so I'll close

closed 2019-08-20 by larsoner