mne-tools/mne-python #2482: website what's new is behind

dgwakeman, 2015-09-21:
http://martinos.org/mne/stable/whats_new.html

No 0.9 info and current information is old

Eric89GXL, 2015-09-21:
The 0.9 is missing, but I'm not sure what you mean about the current information being old. You need to look at the development version for that, if I understand what you're talking about:

http://martinos.org/mne/dev/whats_new.html

Eric89GXL, 2015-09-21:
This should be auto-fixed once we release 0.10 in the next week or so, but I'll leave this open as a reminder to make sure it doesn't say "current" at the top of the page when releasing

Eric89GXL, 2015-09-21:
I updated the release instructions to include updating the version number in `whats_new.py`, that should fix it for the future:

https://github.com/mne-tools/mne-python/wiki/How-to-make-a-release

Thanks for the report.

closed 2015-09-21 by Eric89GXL

dgwakeman, 2015-10-23:
opening this again @Eric89GXL, or do we need to wait for a website rebuild overnight? for the stable version

agramfort, 2015-10-23:
yes we need to update the web site for new release.

the way to go is to update https://github.com/mne-tools/mne-tools.github.io
and then I git pull at MGH

Eric89GXL, 2015-10-25:
I can't do it at the moment because Mayavi is misbehaving on my system.