mne-tools/mne-python #2489: bug in epochs.plot() with single channel

mainakjas, 2015-09-25:
@jaeilepp I discovered this bug:

```py
import mne
from mne import io
from mne.datasets import sample
data_path = sample.data_path()

raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'
event_id, tmin, tmax = 1, -0.2, 0.5

#   Setup for reading the raw data
raw = io.Raw(raw_fname)
events = mne.read_events(event_fname)

# pick EEG channels
picks = mne.pick_types(raw.info, meg=True, eeg=True, stim=False, eog=True,
                       exclude='bads')
# Read epochs
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=(None, 0), reject=dict(eeg=80e-6, eog=150e-6),
                    preload=True)
epochs.pick_channels(['EEG 005'])
epochs.plot()
```

I see a flat channel basically

[image]

jaeilepp, 2015-09-25:
I think it has something to do with reference channel. Does mne use average EEG reference? You can try plotting two channels and marking the other one bad. The other one falls flat immediately.

mainakjas, 2015-09-25:
shouldn't it come as an option in the `proj` button if the average EEG reference should be applied or not? That's what happens with `raw` but not with `epochs`

jaeilepp, 2015-10-02:
The projections are applied automatically. You should construct the epochs with param ``proj=False`` to see the proj button.

mainakjas, 2015-10-02:
hmm ... okay. It wasn't obvious to me. Feel free to close

closed 2015-10-03 by agramfort