mne-tools/mne-python #2490: evoked.plot_topomap auto mode for GFP peaks

agramfort, 2015-09-26:
if would be great if evoked.plot_topomap could automatically set the times of interest eg. by finding the peaks for the global field power.

I would do something like:

evoked.plot_topomap(times='auto')

there might be a bit of work with peak finding though...

jona-sassenhagen, 2015-10-02:
Maybe as simple as

    if times == "auto":
        gfp = evoked.data.std(axis=0)
        times = argrelmax(gfp, order=int(evoked.info["sfreq"] / 5))

I've also played around with find_peaks_cwt a bit, but this one actually works better for e.g. the examples.

jona-sassenhagen, 2015-10-02:
Also maybe the key should be "peaks" and not "auto" because you probably also want an explicit way of calling the dumb sequence (like it is right now). Not `None | "auto" | array`, but `"auto" | "peaks" | array`
And make it default to `auto` and produce the current behavior.

agramfort, 2015-10-02:
I like that

Thanks for the suggestion

jona-sassenhagen, 2015-10-02:
Should I make a PR?

agramfort, 2015-10-02:
Go for it if you have some time

jona-sassenhagen, 2015-10-15:
Closed by #2526, #2506?

Eric89GXL, 2015-10-15:
Agreed, @agramfort reopen if we're wrong

closed 2015-10-15 by Eric89GXL