mne-tools/mne-python #2491: click on evoked.plot to get butterfly plot

agramfort, 2015-09-26:
currently one can click on a line of evoked.plot to get channel name.

it would be great to be able to click on an empty region to plot the topomap of this time instant.

A more advanced used case could be to be able to select a time interval in which data is averaged to produce the topomap

agramfort, 2015-09-26:
maybe for @jaeilepp

dengemann, 2015-09-26:
> A more advanced used case could be to be able to select a time interval in which data is averaged to produce the topomap

I like this! +1

Eric89GXL, 2015-09-26:
Sounds good, but maybe trigger it by something other than a left click in
empty space. It's pretty easy to accidentally click in space when trying to
click a line/channel, and it would be annoying to have a plot pop up due to
such a minor click miss. So maybe middle or right click instead.

agramfort, 2015-09-26:
fine with me

closed 2015-10-21 by jaeilepp