mne-tools/mne-python #2497: Bug mne.cov.compute_covariance + multiprocessing

LaetitiaG, 2015-09-29:
Hi,
I just want to mention that I used the mne.cov.compute_covariance function and it works with n_jobs =1 (as in the example) but if I put more jobs, I have the following error message:

File "/usr/lib/python2.7/multiprocessing/pool.py", line 558, in get
raise self._value

PicklingError: Can't pickle <class 'mne.cov._ShrunkCovariance'>: it's not found as mne.cov._ShrunkCovariance

dengemann, 2015-09-29:
> PicklingError: Can't pickle : it's not found as mne.cov._ShrunkCovariance

Ah that's annoying. We did not have that before. Cannot take a look right
now. We should bisect and see when this came up. Laetita btw. you can use
MKL + Anaconda. In my experience it will be faster than using joblib here.

agramfort, 2015-09-29:
denis can you reproduce?

Eric89GXL, 2017-02-16:
I'm going to go ahead and close this as an isolated upstream bug (probably `multiprocessing`), but if it comes back up or is still happening, let's reopen

commit 6e980003 (2017-10-03): [FIX/MAINT] fix #2497 and #4140 n_jobs in cov (#4632)

* fix #2497 and #4140

* fix pep8

* satisfy points raised by @agramfort
closed 2017-02-16 by Eric89GXL