mne-tools/mne-python #2502: FIX: Check magnetometer type in SSS

wronk, 2015-10-01:
The geometry of the sensors is important for SSS processing. At least at UW, our MEG marks gradiometer with the wrong geometry type (T2 instead of the T3). The Elekta coil geometry is only defined for T3 right now, so the SSS functions should check that that the magnetometers are T3, which I'll probably add in the tSSS PR #2400. If they're not T3, we might want to warn the user and then prompt them to use a (yet to be written) function that will update the mag types if appropriate.

Eric89GXL, 2015-10-01:
I can write a little function tomorrow to allow us to update it.

Eric89GXL, 2015-10-01:
Assuming #2503 gets merged quickly, @wronk you rebase #2374 and add a check there. Basically if fine cal is not used, and T1/T2 type magnetometers are detected, give a `warnings.warn(...)` that they probably need to use `fix_mag_coil_types` on their data.

wronk, 2015-10-19:
This is addressed now. If any T1/T2 channels are present when `maxwell_filter` is called (regardless of fine-cal or other params), they are warned to use `fix_mag_coil_types`.

Should we wait to close until the (oversized) fine-cal PR is merged?

Eric89GXL, 2015-10-19:
Yeah this should auto-close

PR #2374 (2015-10-27): [MRG] SSS: Add Fine Calibration and Cross-talk Correction
Add improved calibration to the SSS functionality. This should increase the SSS shielding factor by ~10x making it useful for actual data processing.

* [x] Add option to use sensor geometry from `sss_cal.dat` files instead of `raw`
* [x] Quantify difference in geometry and send to `logger` along with fine calibration file used
* [x] Add point-like magnetometers (X, Y, and Z) to gradiometer calcs
* [x] Add sensor imbalance weights to gradiometer calcs and calibration coeffs to magnetometer calcs
* [x] Add cross-talk correction
* [x] Add [... cut]
closed 2015-10-27 by agramfort