mne-tools/mne-python #2505: MNE logos?

choldgraf, 2015-10-01:
I'm putting together a poster for SfN, and was hoping to put some of the logos of open-sources packages that I use on the poster. I'm not sure if it'll look alright, but I want to give it a shot because free and powerful software doesn't get enough love in the world.

Does MNE have a "preferred" image that people should use when they are visually referencing the package? Any place to find SVG files or something? This might be a good place to show off that new logo :)

kingjr, 2015-10-02:
How about this?
https://github.com/mne-tools/mne-python/blob/d42a711f5184f64a251398bc39834cf46ac60d17/doc/generate_mne_logos.py

choldgraf, 2015-10-02:
That would be great - is this ready for prime-time, so to speak? I don't want to put anything on a poster before it's finalized. I know there was a vote a few months back...

kingjr, 2015-10-02:
I would suggest you do it, and get feedback, and review the generating code accordingly ;)

choldgraf, 2015-10-02:
hehe, ok, I will do so

dengemann, 2015-10-02:
At least it's aleady on our new website. So go for it and yeah get some
feedback!

choldgraf, 2015-10-12:
FYI the code to generate logo works fine for me, though it would be nice if we could export an SVG as well. What if there were a `generate_mne_logo.py` script that asked you to input the file extension, save path, and maybe extensions. And then there could be a separate file that calls this in a particular way for use in _static. Thoughts?

choldgraf, 2015-10-12:
(note, SVG works fine for me when I just hard-code the file extension into the script)

Eric89GXL, 2015-10-12:
-1 on prompts, optparse (or whatever we use) is well suited to handling
sensible defaults and command line options though

choldgraf, 2015-10-12:
Oh ya, I agree, I meant positional args not a prompt. I hate prompts :-)

Eric89GXL, 2015-10-13:
PR welcome :) Please use `get_optparser` instead of using positional arguments, though, it allows more flexibility

choldgraf, 2015-10-13:
cool - I'll give this a shot...sometime :)  probably after SfN

choldgraf, 2015-10-23:
Hey guys - so the logo creation script worked great for me and I'm closing this issue. I put a bunch of python logos at the bottom-left of my poster, and during the presentation this guy walked and said he worked for the Mathworks. He asked a bunch of leading questions about how it was possible for me to do these analyses in python, if there was anything I wish python accomplished that it didn't do now, how I learned python, etc. Looks like they're doing some market analysis at SfN...kinda weird :P

closed 2015-10-23 by choldgraf

dengemann, 2015-10-23:
> He asked a bunch of leading questions about how it was possible for me to do these analyses in python, if there was anything I wish python accomplished that it didn't do now, how I learned python, etc. Looks like they're doing some market analysis at SfN...kinda weird :P

What? Tell us more about it :)

choldgraf, 2015-10-23:
I mean he was basically just trying to figure out some details about how I did encoding model stuff in python, what packages I used, how I learned how to use them, and whether I thought it was all I needed for my analysis. Oh and he also wanted to make sure that I knew that Matlab was free for people at Berkeley since we have a site license. I tried to be as friendly as possible, since I didn't want to go on a Matlab rampage in the middle of my poster presentation w/ a bunch of people standing there, but it was a strange interaction for sure. He said something like "well these days we just have to accept that Matlab isn't the only game in town anymore" and walked away.

dengemann, 2015-10-23:
Well done Chris. Very interesting interaction indeed ... what a pitty I
wasn't there.

Eric89GXL, 2015-10-24:
Yes thanks for being an excellent implicit advertisement / ambassador for
Python :)

choldgraf, 2015-10-24:
Haha, I just feel like people who spend their time doing free and open
software development don't get enough credit for their work. Not trying to
threaten Mathworks though apparently that's what I did :)