mne-tools/mne-python #2507: load_data problem with BrainVision

teonlamont, 2015-10-03:
Because the brainvision module wasn't modified for the new  `_read_segment`, and `_read_segment_file` methods, it errors with withe `raw.load_data` method.

To reproduce the error:

```python
from mne.io.brainvision.tests import test_brainvision
from mne.io import read_raw_brainvision

raw = read_raw_brainvision(test_brainvision.vhdr_path, preload=False)
raw.load_data()
```

teonlamont, 2015-10-03:
thanks for h/t on the bug @reddigari

Eric89GXL, 2015-10-03:
I can probably fix this on Monday unless there is someone more familiar with the Brainvision code who wants to tackle it.

teonlamont, 2015-10-03:
might be able to do it tomorrow. what is the general way the two methods work together. I might as well fix the bci module I'm adding to handle this properly.

Eric89GXL, 2015-10-03:
All you should need to do is provide the `_read_raw_segment` function. See how it's done in `fiff/raw.py`, basically given a set of time indices you need to return the right chunk of data.

Eric89GXL, 2015-10-03:
(check `base.py` as well, if this isn't made clear in the base class there, then please improve the docs so the next person who wants to do this can do it more easily)

Eric89GXL, 2015-10-15:
Fixed by #2517

closed 2015-10-15 by Eric89GXL