mne-tools/mne-python #2519: IOError when plotting source locations

bam54, 2015-10-07:
I'm trying to run plot_spm_faces_dataset.py on Ubuntu 14 with Python 2.7.6.  At the end of the script, stc.plot(hemi='both', clim='auto') is called.  At first, an error was thrown in io.py when trying to initialize a Surface instance: it looked like the program was looking in os.environ for a value (SUBJECTS_DIR) defined earlier in plot_spm_faces_dataset.py that hadn't been added, so I simply added this value.  After this, another error was thrown from io.py in read_geometry: 

    IOError: [Errno 2] No such file or directory: u'~/mne_data/MNE-spm-face/subjects/spm/surf/both.inflated'

Is this file supposed to be located on my system or is there a bug somewhere?

Eric89GXL, 2015-10-08:
Try upgrading `PySurfer`, I suspect you are using an outdated version

agramfort, 2015-10-08:
and yes these files should be on your system.

bam54, 2015-10-08:
Upgrading PySurfer did the trick, I was using 0.4.  Thank you!

closed 2015-10-08 by agramfort