mne-tools/mne-python #2520: ValueError: You have 1 events shorter than the shortest_event.

Sarwat-Fatima, 2015-10-08:
I am analyzing the raw data for MEG/EEG. It gives me following warning but I am not sure what does it mean. Also the visualization of the data shows me large blue box after a few seconds. It can be seen in the image. Can someone tell me why it is showing me this? 

"ValueError: You have 1 events shorter than the shortest_event. These are very unusual and you may want to set min_duration to a larger value e.g. x / raw.info['sfreq']. Where x = 1 sample shorter than the shortest event length."

[image]

jaeilepp, 2015-10-08:
I think the error means that you have a trigger value that is the length of one sample. This is probably because of overlapping triggers (response pads on top of stimulus for example). You should probably find out what causes this. You can use param ``shortest_event=1`` for ``find_events`` to include the event.
Are you using continuous movement compensation? Did you maxfilter your data with mc on? To me that looks like the signal from the coils.

Sarwat-Fatima, 2015-10-08:
So far I have just followed the tutorial at http://martinos.org/mne/stable/python_tutorial.html 
I have two versions of the data, one raw and one max-filtered data. If I plot the maxfiltered one, the blue box becomes considerably small in width as in the figure below. But in raw data it goes till the end. 
People who collected this data explained the max-filtered process as, 
1) Signal Space Separation to remove enviornmental noise
2) detection of bad channels 
3) notch filtering of 50 Hz line noise, its harmonics and of the HPI  coil signals
4) compensating for movement every 10 ms within each run
5) aligning the data across runs to match the head position 

[image]

dgwakeman, 2015-10-08:
Hi Sarwat,
This error is meant to indicate that one of your trigger codes is only one sample wrong. This is likely caused by a Neuromag acquisition bug, which occasionally turns off one of the binary channels/digits before the others, in multiple binary channel triggers. You likely do NOT want to use `shortest_event=1`. As that will include this as a separate trial. You can run that once if you would like to then use the output to find where this problem occurs, and take a closer look at the trigger channel in that time window. Maxfilter should play no role in this.

Assuming I am correct and it is this binary stepping problem, you simply want to use a larger value for min_duration.

Also, feel free to ask questions like this on the mne_analysis list in the future. We try to keep github for code development.

hth
d

Sarwat-Fatima, 2015-10-08:
Okay. Thankyou. I will use mne_analysis for any further questions.

closed 2015-10-08 by Sarwat-Fatima

dengemann, 2015-10-08:
Hi there.
This problem can also occur when you use event defitions from the raw data
but resample your data later. It can mess with your trigger channel.