mne-tools/mne-python #2523: no grand_average for TFRs

jona-sassenhagen, 2015-10-11:
There is a function to average a list of evokeds, but not for AverageTFRs (right?).

Should be fairly simple.

jaeilepp, 2015-10-15:
> Should be fairly simple.

Except the data needs to be interpolated for bad channels.

agramfort, 2015-10-15:
just intersect channels as you cannot interpolate properly phase or power.

jona-sassenhagen, 2015-10-15:
@agramfort how about an option to either intersect, or just average over the good subset?

agramfort, 2015-10-15:
>
> just average over the good subset?
>
that's what I mean by intersect (the good ones)

Eric89GXL, 2015-10-15:
@agramfort I think the "subset" refers to subset of subjects with the given channel good, i.e. for each channel, you give the average over the subjects for which the channel was good. I guess that means each channel could have a different subset of subjects averaged over. I guess it's okay with me as an option, so long as it's not the default. People should only see that one if they know the number of averages and subset of subjects could differ channel-by-channel.

agramfort, 2015-10-15:
ok agreed

jona-sassenhagen, 2015-10-15:
Yes that's what I meant.

jona-sassenhagen, 2015-10-16:
I have an implementation. It's seemingly actually pretty straight forward (at least the intersection solution). I'll PR once any other of my current PRs is closed.

jaeilepp, 2015-10-16:
I also have something, but it's not that straight forward. It would be interesting to compare the implementations.

jona-sassenhagen, 2015-10-16:
Here is mine:

```
def grand_average(all_tfr, drop_bads=True):
    """Make grand average of a list TFR data

    Selects the subset of good channels across all datasets.

    The grand_average.nave attribute will be equal the number
    of datasets used to calculate the grand average.

    Note: Grand average TFR shall not be used for source localization.

    Parameters
    ----------
    all_evoked : list of AllTFR data
        The TFR datasets.
    drop_bads : bool
        If True, bad MEG and EEG channels are dropped.

    Returns
    -------
    grand_average : AllTFR
        The grand average data.

    Notes
    -----
    .. versionadded:: 0.10.0
    """
    # check if all elements in the given list are evoked data
    if not all(isinstance(e, AverageTFR) for e in all_tfr):
        raise ValueError("Not all the elements in list are AllTFR data")

    # Copy channels to leave the original evoked datasets intact.
    all_tfr = [e.copy() for e in all_tfr]

    # Interpolates if necessary
    if drop_bads:
        for tfr in all_tfr:
            tfr.drop_channels(tfr.info['bads'], copy=False)
    else:
        pass
        # implement good channel subset here somehow??

    equalize_channels(all_tfr)  # apply equalize_channels
    # make grand_average object using combine_evoked
    grand_average = combine_tfr(all_tfr)
    # change the grand_average.nave to the number of Evokeds
    grand_average.nave = len(all_tfr)
    # change comment field
    grand_average.comment = "Grand average (n = %d)" % grand_average.nave
    return grand_average

def combine_evoked(all_tfr, weights='nave'):
    """Merge TFR data

    Data should have the same channels and the same time instants.
    Subtraction can be performed by passing negative weights (e.g., [1, -1]).

    Parameters
    ----------
    all_tfr : list of TFR
        The evoked datasets.
    weights : list of float | str
        The weights to apply to the data of each TFR instance.
        Can also be ``'nave'`` to weight according to tfr.nave,
        or ``"equal"`` to use equal weighting (each weighted as ``1/N``).

    Returns
    -------
    evoked : AllTFR
        The new TFR data.

    Notes
    -----
    .. versionadded:: 0.10.0
    """
    tfr = all_tfr[0].copy()
    if isinstance(weights, string_types):
        if weights not in ('nave', 'equal'):
            raise ValueError('Weights must be a list of float, or "nave" or '
                             '"equal"')
        if weights == 'nave':
            weights = np.array([e.nave for e in all_tfr], float)
            weights /= weights.sum()
        else:  # == 'equal'
            weights = [1. / len(all_tfr)] * len(all_tfr)
    weights = np.array(weights, float)
    if weights.ndim != 1 or weights.size != len(all_tfr):
        raise ValueError('Weights must be the same size as all_tfr')

    ch_names = tfr.ch_names
    for e in all_tfr[1:]:
        assert e.ch_names == ch_names, ValueError("%s and %s do not contain "
                                                  "the same channels"
                                                  % (tfr, e))
        assert np.max(np.abs(e.times - evoked.times)) < 1e-7, \
            ValueError("%s and %s do not contain the same time instants"
                       % (evoked, e))

    # use union of bad channels
    bads = list(set(tfr.info['bads']).union(*(ev.info['bads']
                                                 for ev in all_tfr[1:])))
    tfr.info['bads'] = bads

    tfr.data = sum(w * e.data for w, e in zip(weights, all_tfr))
    tfr.nave = max(int(1. / sum(w ** 2 / e.nave
                                   for w, e in zip(weights, all_tfr))), 1)
    return tfr
```

I guess yours is a lot better :)

jaeilepp, 2015-10-16:
Hey, that looks pretty good. I wasn't aware of this ``equalize_channels`` function. I did something more stupid. The documentation should be changed that it also accepts ``AverageTFR``. Just fix the typos and make a PR.
Also, I think there's something wrong with the ``plot_topo`` of TFR. I'll look into it.

jaeilepp, 2015-10-16:
And by typos I mean references to evoked object.

jaeilepp, 2015-10-16:
And in more general, should the ``weights`` parameter be exposed to the user? With this implementation, the channel equalization would have to be made by the user before calling the ``combine_tfr``, if he/she wants to use ``weights='equal'``. Same goes for evoked grand average.

jona-sassenhagen, 2015-10-16:
I had actually written a setup all on my own to do the job of equalize_channels before I found out the function exists ...

jona-sassenhagen, 2015-10-16:
>And in more general, should the weights parameter be exposed to the user? With this implementation, the channel equalization would have to be made by the user before calling the combine_tfr, if he/she wants to use weights='equal'. Same goes for evoked grand average.
Right now I'd tend to implement the function as similar to the one for evokeds as possible, and if we want to change evokeds later, that's a different story. Thoughts?

jona-sassenhagen, 2015-11-10:
Closed by #2569

closed 2015-11-10 by jona-sassenhagen