mne-tools/mne-python #2533: ENH: spatial color butterfly plots

jona-sassenhagen, 2015-10-16:
This is how I plot:
[image]

- butterfly style
- channels are color-coded by mapping their x/y/z positions into RGB space (see legend in plot)
- combine butterfly and topomaps, EEGLAB style

I think this is a very good way of visualizing spatio-temporal dynamics. I've discussed this with @dengemann before and IIRC he liked it.

Color coding is fairly straight-forward and I'd like to begin with that. The code I used above is simply:
```
                if spatial_colors:
                    if any([i in bad_ch_idx for i in idx]):
                        raise ValueError('Spatial colors are incompatible with "bads".')
                    try:
                        chs = [evoked.info['chs'][i] for i in idx]
                    except:
                        raise ValueError("No valid channel locations found.")
                    chspace = np.asarray([list(chs[ch]['loc'][0:3]) 
                                          for ch in range(len(chs))])
                    chspace -= chspace.min()
                    chspace /= (chspace.max()/1)
                    ax._get_lines.color_cycle = iter(chspace)
```

in `_plot_evoked`. Can I implement this? My PR schedule is comparatively empty right now.
(Topomaps and the in-plot legend are a bit more complicated, but this part is fairly simple.)

agramfort, 2015-10-16:
nice. why not

I am sure @kingjr will like this :)

Eric89GXL, 2015-10-16:
Sure. I wonder if we could do something more systematic with the colors, though, like have it go blue on the left to red on the right, with value from light at the top to dark at the bottom. That way, given a color in the plot, you immediately have a sense of where it is without having to look at the key.

jona-sassenhagen, 2015-10-16:
@dengemann said we should maybe find a different color map.

kingjr, 2015-10-16:
yes I really like the idea. I would do a 2D transform though, as the sensors are on a surface, not a volume. Have you tried using a double gradient : e.g. red/blue for coding left-right and black/white for front-back?

jona-sassenhagen, 2015-10-16:
@kingjr @Eric89GXL haha

Eric89GXL, 2015-10-16:
@jona-sassenhagen there you go, red/blue light/dark wins sounds like :)

kingjr, 2015-10-16:
great minds :]

jona-sassenhagen, 2015-11-26:
closed for #2638

closed 2015-11-26 by jona-sassenhagen