mne-tools/mne-python #2535: BUG: TimeDecoding

kingjr, 2015-10-16:
There's a bug in `TimeDecoding` when `n_jobs=-1`. Please wait before release.

dengemann, 2015-10-16:
Confirmed, I remember we had this in BA with n_jobs=2.

kingjr, 2015-10-16:
It's because of the way I chunk the data to avoid the memory duplicates. 

However, the preindexing option you mentioned last time should eventually get rid of this kind of issues, and greatly clarify the code. 

I'm on this in the meantime :/

dengemann, 2015-10-16:
while you are at it checkout the line_rofiler -- it will make you understand.
Also checkout the first steps we took with @fraimondo : https://github.com/fraimondo/mne-python/commit/973c7d5623cfa3d0ff143af5861fa826482e7ef6

dengemann, 2015-10-16:
... and the 2 previous commits.

kingjr, 2015-10-16:
Ok, let's take things in order. First fix, then improve in next PR?

When fixing I realized that the predictions cannot be currently parallelized in `TimeDecoding` (because the parallel runs across testing time points).

dengemann, 2015-10-16:
Yes, sounds good. 

> When fixing I realized that the predictions cannot be currently parallelized in TimeDecoding (because the parallel runs across testing time points).

Refactoring is keeping you busy ;)

kingjr, 2015-10-16:
> Refactoring is keeping you busy ;)

oh dear, a never ending story

closed 2015-10-16 by kingjr