mne-tools/mne-python #2548: EHN: #Easy add pick_type to epochs.average()

kingjr, 2015-10-21:
If anyone have some free time..

agramfort, 2015-10-21:
I don't get it. What is it you cannot do currently?

kingjr, 2015-10-21:
we can do `epochs.average(picks=picks)` but not `epochs.average(pick_types='eeg')`.

Also, IIRC, you can't do `epochs.average()` for `seeg` data (because seeg is not part of the default picks)

agramfort, 2015-10-21:
> we can do epochs.average(picks=picks) but not epochs.average(pick_types='eeg').

write one more line to get the picks or do

epochs.pick_types(meg=False, eeg=True).average()

> Also, IIRC, you can't do epochs.average() for seeg data (because seeg is not part of the default picks)

this is a bug. PR welcome

kingjr, 2015-10-21:
> write one more line to get the picks or do 
> epochs.pick_types(meg=False, eeg=True).average()

Sure, my feeling was that it'd be more consistent to have a pick_types param if you have a picks. But I won't fight for this.

> this is a bug. PR welcome
all right then

closed 2015-10-21 by agramfort