mne-tools/mne-python #2553: IAS channels

jaeilepp, 2015-10-23:
MEG data measured with Triux system has the IAS channels at the beginning of the ``ch_names`` before the data channels. This messes up the iterator indexing in ``mne.viz.topo.iter_topography``, so that for instance example ``plot_topo_customized.py`` throws an ``IndexError``. It could be fixed in ``iter_topography``, but I feel the channel ordering should be different in the info structure itself.
wdyt?

Eric89GXL, 2015-10-23:
Does this mean that we'll need our maxwell filtering algorithm to change channel names on filtering? Do you know if that's what MaxFilter does? grrr...

In any case, I think we already have a name-sanitizing function that removes or adds spaces, e.g. treats "MEG 0113" and "MEG0113" as equivalent. Perhaps that will help here.

jaeilepp, 2015-10-23:
I don't think MaxFilter reorders or renames the channels, if that's what you were asking.
The issue here is that ``iter_topography`` returns the index of the channel in ch_names, whereas the example uses indexes from 0. The user could of course modify the info structure himself, but this seems hackish. And I'm not sure if this is the only place that is "broken" because of it.
To me it just seems counter intuitive that the IAS channels are before the data channels. Is that somehow in the fif file structure or can we change it?

Eric89GXL, 2015-10-23:
Ahh, I see what you mean now, I was confused. I thought they appended IAS to channel names, not that the IAS data channels came before the data channels.

Actually none of our code should depend on the channels being in a certain order, e.g. data before IAS, or MEG before EEG. In theory, channels should be able to be reordered randomly, and analysis should proceed from start to finish without issue. Places where that doesn't happen are bugs. In fact we should probably add a test for this...

So the short answer is that, no, we shouldn't reorder the channels and/or info, we should fix our code to properly accommodate arbitrary channel orderings.

jaeilepp, 2015-10-23:
Yeah, it's not a biggie, since the IAS channels probably never get picked past the raw stage anyway. So perhaps it's okay to exclude the IAS channels in ``iter_topography``?

Eric89GXL, 2015-10-23:
I think `iter_topography` should only iterate over data channels, yeah. So making it use the right `pick` functions in a way that doesn't care about the ordering of said channels would be the way to go.

closed 2015-10-29 by agramfort