mne-tools/mne-python #2555: Visualizing brains in 3-D space?

choldgraf, 2015-10-23:
I'm trying to figure out if it's feasible to visualize brains in 3-D space with python. I know there's a lot of software out there (e.g., freesurfer/pysurfer, or pycortex) that lets you do this with solid MR scans, but in my case I basically just have a Nifty file with pretty low-quality scans.

Is there any way in MNE (or python more generally) to just load a Nifty file of someone's brain reconstruction, a 3-D array of numbers denoting electrode positions, and then make scatterplots etc on top?

agramfort, 2015-10-25:
the best way to visualize brain slices is with nilearn AFAIK.

we use it in
http://martinos.org/mne/dev/auto_examples/inverse/plot_lcmv_beamformer_volume.html#sphx-glr-auto-examples-inverse-plot-lcmv-beamformer-volume-py

if you need more you'll need to code a bit.

choldgraf, 2015-10-25:
That looks useful - thanks for the tip. I basically want to do things like
what you just linked to, only on a 3D brain w/ some orientation. There are
a few packages that do this in matlab but I'm hoping something exists in
python for this. Unfortunately the ECoG world is mostly a matlab world :P

agramfort, 2015-10-25:
sounds like a good sunday hack :)

choldgraf, 2015-10-25:
Hehe, something I am happy to assist somebody with if they want to do some co-hacking with me, but I have about 0 experience doing 3D plotting. Right now I just take a 2D snapshot of the brain, and use the mouse to click the X/Y coordinates of each electrode on that snapshot (which is stupid, I know)

agramfort, 2015-10-25:
depends how often you need to do this :)

choldgraf, 2015-10-25:
Well, at least once for every subject, but it means that if you ever want to change something about that picture (e.g., shrink it or blow it up, shift it, change the angle, etc) you have to redo the entire thing. Plus it just seems silly to take a full 3D representation of a brain, along with 3D electrode positions, and then throw all of that away in lieu of manually clicking an image.

choldgraf, 2015-10-25:
I'm gonna ask the nibabel / nipy folks if they have any experience with Mayavi. I actually got this sort-of working, and if it becomes something stable then I'm happy to contribute it to MNE (e.g., some sort of high-level function that accepts a volumetric representation of a brain + x/y/z positions of electrodes). If you know of anyone with nibabel and/or mayavi experience, I'd love to shoot them questions :)

Eric89GXL, 2017-02-16:
I'm closing this one, as I think we have made progress on some fronts and opened more specific issues on others. @choldgraf if there are things in here that aren't covered elsewhere, please reopen and retitle

closed 2017-02-16 by Eric89GXL

choldgraf, 2017-02-16:
yes I think so - plenty of issues to be opened, but I think those are all on the Mayavi repo ;-)