mne-tools/mne-python #2557: auto build of dev doc

agramfort, 2015-10-24:
we should have a nightly build of the dev doc.

sklearn is planning to use circle CI https://github.com/scikit-learn/scikit-learn/pull/5578

any other opinion?

agramfort, 2015-10-24:
cc @jaeilepp

Eric89GXL, 2015-12-21:
We can probably do it with CircleCI and I trieid a couple months ago, but it would be a lot easier if they would disable autodetection of type. See issue I opened here:

https://discuss.circleci.com/t/disable-autodetection-of-project-or-application-of-python-venv/235

If they don't move on it in a couple months, I might try using system Python, but suspect it will be too difficult for the effort given how many dependencies we have.

jasmainak, 2015-12-21:
yeah, this would be great to have and it would also encourage more contributions to the docs.

jasmainak, 2015-12-21:
By the way, it would be cool to have a way to build only one file / directory in the examples selectively. Currently, I do it by moving the folders outside the examples/ dir except the one I need to process (it's a bit less involved as I know sphinx processes the files alphabetically). And then move it back once it's built. It's useful for cross-linking the examples and examples images ...

Eric89GXL, 2016-03-16:
@jaeilepp has done this now. @agramfort do we need to do something to trigger an upload from `circleci` to `github`?

agramfort, 2016-03-16:
>
> @jaeilepp <https://github.com/jaeilepp> is working on it.
>
I am dreaming of a working nightly build of the doc ...

Eric89GXL, 2016-03-16:
Oh right, it dosen't quite work fully yet. I'll leave this open. Do we have a plan for how to upload it once the CircleCI artifact is correct (i.e., once Mayavi works)?

agramfort, 2016-03-16:
yes mayavi is the "only" pb ...

Eric89GXL, 2016-03-27:
I think this one is good, @jaeilepp feel free to reopen if I'm wrong

closed 2016-03-27 by Eric89GXL