mne-tools/mne-python #2563: Cluster Statistics Multi cluster problem

ImmanuelSamuel, 2015-10-26:
I have the following topo and a number of observations of the same
[image]

```
T_obs, clusters, cluster_pv, H0  = mne.stats.permutation_cluster_test(X, threshold=None, n_permutations=1024, tail=0,
                                   connectivity=connectivity, verbose=None, n_jobs=20, seed=None, max_step=1, exclude=None, 
                                   step_down_p=0, t_power=1, out_type='mask', check_disjoint=False, buffer_size=1000)
```
This gives me only one cluster but from the topo we can see that there are two clusters.
What am I doing wrong here?
[image]

ImmanuelSamuel, 2015-10-26:
Is there a way to set max cluster size?

Eric89GXL, 2015-10-27:
I wonder if the sensors on the right are close enough to be considered "neighbors" by the algorithm...?

No way to set max cluster size, but changing the preliminary threshold to be more restrictive will effectively shrink the cluster sizes because it will reduce the number of clusterable points.

ImmanuelSamuel, 2015-10-27:
Interesting. Do you know where was the algorithm obtained from?

Eric89GXL, 2015-10-28:
It's a general algorithm, see e.g.  Maris/Oostenveld (2007), "Nonparametric statistical testing of EEG- and MEG-data" Journal of Neuroscience Methods, Vol. 164, No. 1., pp. 177-190. doi:10.1016/j.jneumeth.2007.03.024. You can set the connectivity however you like, can't remember who did our specific implementation for sensor space. You could try using `git blame` (or GitHub's equivalent) to figure it out if you're really curious.

Eric89GXL, 2017-02-16:
I'm going to close this since the method seems to work properly as implemented.

closed 2017-02-16 by Eric89GXL