mne-tools/mne-python #2574: ENH: avoid magic file reading in plotting functions if desired (e.g. plot_trans)

dengemann, 2015-10-28:
Debugging would be easier if you could directly specfiy the path to the surfaces, etc. I don't like this implicit guessing of file names, especially not when debugging.

agramfort, 2015-10-28:
+1

I've had the problem in the past.  It's the magic used in mne_analyze too.

Eric89GXL, 2015-10-28:
I think it does tell you which file it uses in the logger, so that should facilitate debugging. Most of the time it works automatically for people, so I'm -1 on getting rid of the auto-guessing for e.g. head surfaces. But allowing to pass an actual path to a surf or FIF with head BEM sounds fine to me.

dengemann, 2015-10-28:
> I'm -1 on getting rid of the auto-guessing for e.g. head surfaces. But allowing to pass an actual path to a surf or FIF with head BEM sounds fine to me.

that was the idea, being able to pass the names

dengemann, 2015-10-28:
Likewise, all plotting functions should accept the objects as parameters. This IO-bound plotting is not nice to debug (and requires unnecessary IO).

Eric89GXL, 2015-10-28:
Recently I refactored this in a few places, e.g. there is an `_ensure_src` function that takes either a `SourceSpaces` or str, and tries to load it to an instance of `SourceSpaces`; also an `_ensure_trans` function that does the same. Sounds like you want to generalize this practice, which sounds good to me.

agramfort, 2015-10-28:
+1 to have an 'auto' or None param to default to black magic

Eric89GXL, 2015-10-28:
I think that's more or less what's done now, you specify the source as being `'head'` or `'bem'` and it looks at those types of files for the right surface.

dengemann, 2015-10-28:
it's getting especially magic if you can read the file outside of the function but not inside ...

dengemann, 2015-10-28:
Ok found it .... we force having then patch_stats == True when reading the head???

Eric89GXL, 2015-10-28:
Not sure why we'd need that. You can try disabling it to see if it breaks some test. Although with try catches in use, who knows if we're actually getting what we expect :(

dengemann, 2015-10-28:
> Not sure why we'd need that. You can try disabling it to see if it breaks some test. Although with try catches in use, who knows if we're actually getting what we expect :(

let's keep an eye on such complexity-inducing practices, we should avoid a) try catch b) use keyword arguments ... what the heck was it again that's 'True'? Ahhh, how silly, I would have seen it before would we have used a keyword argument.

larsoner, 2018-02-23:
I *think* we have removed the magic-trans-searches from the code at this point. The ones that remain are the ones for BEM and head surfaces, which seem to be pretty robust, since they should rely on / reflect the standard FreeSurfer pipelines (i.e., have the same expectation as the existence of `surf/lh.white`). I'll close for now but if people run into examples feel free to reopen and let's fix.

closed 2018-02-23 by larsoner