mne-tools/mne-python #2575: EDF file - "new epoch" marker

lennyvarghese, 2015-10-28:
For BDF files, Bit 17 marks a "new epoch" event, and thus keeps track of when a recording was paused or resumed. (at least, this is the case for Biosemi ActiveTwo systems)

To support this on my working copy of MNE, I've changed line 287 in io/edf/edf.py to "131071" (2**17-1) from "65536".

I don't suppose this change will break anything (although someone correct me if I'm wrong)...and if that is indeed the case, it might be worth considering this change in future versions of MNE.

agramfort, 2015-10-28:
can you send us a pull request?

how is this done eg in EEGLAB or fieldtrip?

lennyvarghese, 2015-10-28:
I'm not sure that this is handled at all in other packages (though it's been a while since I've used each)...I never knew that this was a feature in BDF files until I was playing with "EDFBrowser" (http://www.teuniz.net/edfbrowser/) and looking at the raw event channel signals.

teonlamont, 2015-11-10:
closed via #2576

closed 2015-11-10 by teonlamont