mne-tools/mne-python #2579: Enlarge ERP plots in plot_evoked_topo

cle1109, 2015-10-29:
I was wondering if it is possible to enlarge the ERP plots in `plot_evoked_topo` (and thereby reducing the space between the single plots). Right now, there's a lot of black space between my plots, which are really small and difficult to interpret.

agramfort, 2015-10-29:
I agree that our plot topo overfits the neuromag setup and is not great
besides for data exploration (not good for paper figures)

jona-sassenhagen, 2015-10-29:
I used to have code for that, but wasn't good enough of a coder to actually finish it. I probably will finish that eventually, but I can't promise when.

cle1109, 2015-11-25:
OK, thanks for the heads up - I'm closing this for now since there is a clear answer to my question.

closed 2015-11-25 by cle1109