mne-tools/mne-python #2581: raise self.failureException('<bound method Kit2FiffModel.<lambda> of <mne.gui._kit2fiff_gui.Kit2FiffModel object at 0x7f9e4df0cd10>> is not false')

ignatenkobrain, 2015-10-30:
```python
======================================================================
FAIL: Test CombineMarkersModel Traits Model
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/builddir/build/BUILD/mne-python-0.10/mne/utils.py", line 633, in dec
    return function(*args, **kwargs)
  File "/builddir/build/BUILD/mne-python-0.10/mne/gui/tests/test_kit2fiff_gui.py", line 35, in test_kit2fiff_model
    assert_false(model.can_save)
AssertionError: <bound method Kit2FiffModel.<lambda> of <mne.gui._kit2fiff_gui.Kit2FiffModel object at 0x7f9e4df0cd10>> is not false
    '<bound method Kit2FiffModel.<lambda> of <mne.gui._kit2fiff_gui.Kit2FiffModel object at 0x7f9e4df0cd10>> is not false' = self._formatMessage('<bound method Kit2FiffModel.<lambda> of <mne.gui._kit2fiff_gui.Kit2FiffModel object at 0x7f9e4df0cd10>> is not false', "%s is not false" % safe_repr(expr))
>>  raise self.failureException('<bound method Kit2FiffModel.<lambda> of <mne.gui._kit2fiff_gui.Kit2FiffModel object at 0x7f9e4df0cd10>> is not false')
```

agramfort, 2015-10-31:
ping @christianbrodbeck @teonlamont

christianbrodbeck, 2015-10-31:
OS, Traits version?

ignatenkobrain, 2015-10-31:
Fedora 24 (Rawhide), 4.5.0

christianbrodbeck, 2015-10-31:
I can't reproduce it on OS X and have no access to Fedora.

ignatenkobrain, 2015-10-31:
I can give you ssh to fedora after few hours.

christianbrodbeck, 2015-10-31:
Is somebody actually wanting to use kit2fiff with Fedora? Because the GUI probably has to be tested independently for platform dependent issues.

Eric89GXL, 2017-03-21:
@ignatenkobrain this hasn't seen action in a while. If you still have the problem with no workaround let us know. In the meantime I suspect the bug might be fixed by upstream libraries (e.g., tvtk).

closed 2017-03-21 by Eric89GXL