mne-tools/mne-python #2582: AttributeError: 'function' object has no attribute 'can_save'

ignatenkobrain, 2015-10-30:
```python
======================================================================
ERROR: Test CombineMarkersModel Traits Model
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/builddir/build/BUILD/mne-python-0.10/mne/utils.py", line 633, in dec
    return function(*args, **kwargs)
  File "/builddir/build/BUILD/mne-python-0.10/mne/gui/tests/test_marker_gui.py", line 33, in test_combine_markers_model
    assert_false(model.mrk3.can_save)
AttributeError: 'function' object has no attribute 'can_save'
```

agramfort, 2015-10-31:
how did you come up with this error? coregistration GUI?

ignatenkobrain, 2015-10-31:
@agramfort I just started nosetests when packaged mne-python.

agramfort, 2015-10-31:
certainly a conflict of traits versions. I am running 4.4.0

ignatenkobrain, 2015-10-31:
I'm running 4.5.0 of Traits.

agramfort, 2015-10-31:
that's weird.

can_save attribute is defined in MarkerPoints class.

no idea what's happening.

Eric89GXL, 2017-02-16:
I think this has been fixed (hopefully) through the trail of #2648

closed 2017-02-16 by Eric89GXL