mne-tools/mne-python #2589: Optimize CSP parameters in grid search

cle1109, 2015-11-04:
I was thinking about optimizing `n_components` in a `GridSearchCV` or `RandomizedSearchCV`. However, the required methods `set_params` and `get_params` are missing from the `CSP` class. Would it make sense to add these methods or is this overkill (because from experience we kind of know that the optimal number of components is either 4 or 6)?

agramfort, 2015-11-04:
> I was thinking about optimizing n_components in a GridSearchCV or
> RandomizedSearchCV. However, the required methods set_params and
> get_params are missing from the CSP class. Would it make sense to [...]
would be great to have more sklearn compatible decoding objects.

For CSP I don't know. You know more than me. But if 6 is a good number,
should we make it the default?

cle1109, 2015-11-05:
Either 4 or 6 is a good default choice. Most people use 6 though because many popular papers use this number (e.g. [Blankertz et al., 2008](http://dx.doi.org/10.1109/MSP.2008.4408441)).

Regarding compatibility with `sklearn`, how would you go about that? The required methods are defined in `BaseEstimator` - should I

1. inherit from `sklearn.base.BaseEstimator` (in addition to `TransformerMixin`)
2. copy `sklearn.base.BaseEstimator` and include it in MNE (e.g. in `mne.decoding.base`)?
3. implement the two methods directly in the `CSP` class?

PS: I think even if we don't want to optimize CSP parameters adding these methods is a good idea, because only then is it possible to perform a parameter search with `CSP` as part of a `Pipeline`.

agramfort, 2015-11-05:
>
> Either 4 or 6 is a good default choice. Most people use 6 though because
> many popular papers use this number (e.g. Blankertz et al., 2008
> <http://dx.doi.org/10.1109/MSP.2008.4408441>).
>
ok we have 4 by default. It should be good.

> Regarding compatibility with sklearn, how would you go about that? The
> required methods are defined in BaseEstimator - should I
>
>    1. inherit from sklearn.base.BaseEstimator (in addition to
>    TransformerMixin)
>    2. copy sklearn.base.BaseEstimator and include it in MNE (e.g. in
>    mne.decoding.base)?
>    3. implement the two methods directly in the CSP class?
>
> we don't want a strong dependency on sklearn. Duck typing suggests just to
implement set_ get_params.

>
>
> PS: I think even if we don't want to optimize CSP parameters adding these
> methods is a good idea, because only then is it possible to perform a
> parameter search with CSP as part of a Pipeline.
>
+1

cle1109, 2015-11-05:
> we don't want a strong dependency on sklearn. Duck typing suggests just to implement set_ get_params.

OK, I'll create a PR. One last question: should this implementation be as simple as possible (i.e. only work for the `CSP` class by e.g. hard-coding the parameter names `n_components`, `reg`, and `log`), or should it be generic (i.e. work for all estimators, which I would do by blatantly copying the relevant code from `sklearn` - which however would entail some additional dependencies such as `warnings`, `signature`, and `six`)?

agramfort, 2015-11-05:
for now keep it simple stupid.

closed 2015-11-26 by cle1109