mne-tools/mne-python #2592: ENH: finding bem surfaces

dengemann, 2015-11-05:
is currently very instable,
it assumes certain hardcoded names, not naming pattenms, and expects head coordinates in some functions. I forgot detail but it's recently been a constant mess to work with different. Who wants to do something good while feeling bored should investigate it and simplify things.

dengemann, 2015-11-05:
It's also a pain that we expcect people do do the linking trick as suggested by the old manual. We should not do that. It should just work. I think we should more adopt the principle of least surprise to make our users more happy.

agramfort, 2015-11-05:
I have add similar issues in the past. But how did this pb show up for you?

dengemann, 2015-11-05:
> I have add similar issues in the past. But how did this pb show up for you?

processign somato data

Eric89GXL, 2016-11-18:
There is a more specific discussion in #3406, feel free to write there as specific examples come up

closed 2016-11-18 by Eric89GXL