mne-tools/mne-python #2594: 403 forbidden

ahibahi, 2015-11-05:
Hi,
I've been using MNE-python for a month with no issues.
Suddenly there's an http error when I try to load the sample dataset
I've tried upgrading and downgrading python as well.still no luck.

This is the code,
from mne.datasets import sample
data_path = sample.data_path() //throwing error at this point

[image]

please help me 
thank you very much

agramfort, 2015-11-05:
can you see if you can download the file directly:

https://s3.amazonaws.com/mne-python/datasets/MNE-sample-data-processed.tar.gz

?

ahibahi, 2015-11-05:
Thank you for the reply.
I'm not able to access it directly as well
Is there any other settings that I should do?

[image]

agramfort, 2015-11-05:
@Eric89GXL did you do something?

ahibahi, 2015-11-05:
I tried installing mayavi library which gave me some errors.So I uninstalled python and reinstalled it.
Now its giving me this 403 error.

agramfort, 2015-11-05:
yes I have the same download problem.

Eric89GXL, 2015-11-05:
Yeah I updated the dataset and probably broke it, will fix soon

ahibahi, 2015-11-05:
It would be really helpful if you can notify me once its fixed.
Thank you very much.

Eric89GXL, 2015-11-05:
Should be all better, feel free to reopen if it still doesn't work

closed 2015-11-05 by Eric89GXL