mne-tools/mne-python #2595: possibly generalize corrmaps

teonlamont, 2015-11-05:
Hey @jona-sassenhagen , thanks for implementing Corrmap. I'm about to use it and i'm pretty excited about it. I am wondering if this could be extended to take a list of principal components, same principle of template matching a spatial distribution but instead of ICs, the spatial maps are from PCA. I haven't looked at the code yet but are you correlating the weights of the corresponding row in the mixing matrix?
I am working on a way of comparing principal components across subjects in a meaningful way.

jona-sassenhagen, 2015-11-08:
Indeed, Corrmap correlates mixing matrices across the supplied objects. So if correlations across channel weights is what you're after, CORRMAP can help. It should work with all spatial filter things - PCA, ICA, CSP.

For now, it should be possible to set an ICA object copy's relevant attributes to that of the PCA you're interested in.

In principle, and if we really want to emphasize across-subject stuff in the next release, maybe constructing a general spatial_filter class could be considered.

jona-sassenhagen, 2015-11-10:
Hm ... ICA (in MNE, and generally) usually includes a PCA stage, so a very simple way to get a "PCA object" would be making an ICA object with an identity matrix for .unmixing_matrix, right?

dengemann, 2015-11-10:
>  would be making an ICA object with an identity matrix for .unmixing_matrix, right?

in theory yes. but it would of course be a hack. the clean way would be to implement an abstract base class for linear unsupervised learning models (all models that do decomposition using a wieght matrix) and then subclass it for ICA and make a new one for .... Teon what is it actually, are you talking about SSP?

jona-sassenhagen, 2015-11-10:
Oh, and of course it would be wholly pointless for the present purpose, as CORRMAP only checks the ICA weights.

teonlamont, 2015-11-10:
@dengemann I just want to be able to compare SSPs across subjects.

agramfort, 2017-07-22:
maybe YAGNI :)

closed 2017-07-22 by agramfort