mne-tools/mne-python #2596: GAT with SVC

teonlamont, 2015-11-05:
I'm trying to use the GAT with the classifier equal to a specified SVC. I am running into this error where it is stating I have multiclass labels. I think this is error in the precision where zero is sometimes represented as a super small float (e.g. 2.02566915e-322). Any suggestions?

cc: @kingjr

```python
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
/Applications/packages/E-MEG/scripts/priming_sensor_analysis_jr.py in <module>()
    143                                    train_times=train_times, clf=clf, cv=cv)
    144     gat.fit(epochs, y=y)
--> 145     gat.score(epochs, y=y)
    146     fig = gat.plot(title='GAT Decoding Score on Semantic Priming: '
    147                    'Unprimed vs. Primed')

/Applications/packages/mne-python/mne/decoding/time_gen.pyc in score(self, epochs, y)
    344                        [self.y_pred_[train] for train in split],
    345                        self.scorer_)
--> 346             for split in splits)
    347 
    348         self.scores_ = [score for chunk in scores for score in chunk]

/Applications/packages/mne-python/mne/decoding/time_gen.pyc in <genexpr>((split,))
    344                        [self.y_pred_[train] for train in split],
    345                        self.scorer_)
--> 346             for split in splits)
    347 
    348         self.scores_ = [score for chunk in scores for score in chunk]

/Applications/packages/mne-python/mne/decoding/time_gen.pyc in _score_slices(y_true, list_y_pred, scorer)
    434         for t, this_y_pred in enumerate(y_pred):
    435             # Scores across trials
--> 436             scores.append(scorer(y_true, np.array(this_y_pred)))
    437         scores_list.append(scores)
    438     return scores_list

/Users/teon/anaconda/lib/python2.7/site-packages/sklearn/metrics/classification.pyc in accuracy_score(y_true, y_pred, normalize, sample_weight)
    177 
    178     # Compute accuracy for each possible representation
--> 179     y_type, y_true, y_pred = _check_targets(y_true, y_pred)
    180     if y_type.startswith('multilabel'):
    181         differing_labels = count_nonzero(y_true - y_pred, axis=1)

/Users/teon/anaconda/lib/python2.7/site-packages/sklearn/metrics/classification.pyc in _check_targets(y_true, y_pred)
     82     if len(y_type) > 1:
     83         raise ValueError("Can't handle mix of {0} and {1}"
---> 84                          "".format(type_true, type_pred))
     85 
     86     # We can't have more than one value on y_type => The set is no more needed

ValueError: Can't handle mix of binary and continuous
```

kingjr, 2015-11-05:
can you describe `scorer` and  `y`?

teonlamont, 2015-11-05:
`y` is an array of ones and zeros. and the `scorer` is the default one. I gave the `ShuffleSplit` for `cv`. Reproducing the settings I had before I used the `GAT` class

kingjr, 2015-11-05:
Mmmmh... ShuffleSplit is not an exhaustive CV which creates a bug. 

The `scorer` tries to score across predicted values (1 and 2 in the default example) and initialized values (0). We would probably need to add a fix in the script to only score across predicted trials...

teonlamont, 2015-12-21:
@kingjr did you recently take care of this with #2673?

kingjr, 2015-12-21:
No it doesn't. I've justs started a branch on this, but I got stuck because it gets too messy:
https://github.com/kingjr/mne-python/tree/gat_shuffledsplit

In a nutshell, not only is ShuffleSplit non-exhaustive (some samples may never be get a prediction), but the cv changes every time you run it; so you can't initialize which values will be predicted or not.

There's probably a fix (test whether a random state has been attributed, generate a temporary cv, initialize values from cv_temp and run cv_temp instead of cv), but it's a bit messy.

Feel free to take over the branch. ITMT, I would advise just using a deterministic (and exhaustive) CV. If you really want to do some benchmarking, fold shuffling and all, you can easily do this outside the GAT.

kingjr, 2016-02-17:
closed by https://github.com/mne-tools/mne-python/pull/2856