mne-tools/mne-python #2599: fix filename and directory of inner_skull after watershed

yuval-harpaz, 2015-11-08:
Hi
since now watershed is nicely wrapped in mne  it is an opportunity to stop this issue of creating a link  bem/watershed/inner_skull_surf > bem/inner_skull.surf
I suggest cpoying rather than linking because links may break when moved from folder to folder or from PC to PC.
now I run
shutils.copyfile(subjects_dir + "/" + subject + "/bem/watershed/" + subject +
         "_inner_skull_surface", subjects_dir + "/" + subject + "/bem/inner_skull.surf")

agramfort, 2015-11-08:
fine with me.

PR welcome

dgwakeman, 2015-11-08:
maybe making this copy an option instead of the default (I like symbolic links), it saves unnecessary wastes of disk space, and is less confusing when one has multiple BEM types. For example, if I generate both a watershed and MEFLASH BEM, I can tell from a simple ls -l, which surface I am using here, while if one copies the files you cannot easily tell.

yuval-harpaz, 2015-11-09:
a link is also fine with me if disk space is an issue
or even 'move' for default behaviour if nobody uses the original ones

agramfort, 2015-11-09:
I see one argument for copying:

- windows users that copy freesurfer files to work later from windows.

but I agree that symlinks are useful to know from what bem type file are
coming from.

if there is no consensus or blocker, let's focus on more urgent matters.

my 2c

Eric89GXL, 2016-11-18:
This has been open for a year with no action. I'll go ahead and close but if this is a needed/wanted feature we can come back to it

closed 2016-11-18 by Eric89GXL