mne-tools/mne-python #2600: Unbundle libs

ignatenkobrain, 2015-11-08:
- [ ] `six`
- [ ] `jdcal`
- [ ] `decorator`
- [ ] `h5io`
- [ ] `tempita`

I think we should use system libs for because those are ready for system-installing

ignatenkobrain, 2015-11-08:
I can make patches for you, it will take few hours, but before I want to ensure that you are interested to accept patches.

dengemann, 2015-11-08:
Not sure what you are proposing. To minimize installation issues we have keep them in externals, I don't see any strong reason to change this, it save's us a lot trouble I think.

ignatenkobrain, 2015-11-08:
But when we're talking about packaging into distros (I'm packaging for Fedora) it *causes* troubles because bundled libraries isn't allowed.

agramfort, 2015-11-08:
for debian there are some patches that are applied to remove these from the
repo and use system versions. I think you should do the same for fedora.

See :

https://packages.debian.org/sid/python/python-mne

agramfort, 2015-11-08:
see

http://anonscm.debian.org/cgit/debian-med/python-mne.git

ignatenkobrain, 2015-11-09:
For Fedora I did unbundling, but I'd want to send to upstream. That's why I created this issue and ask developers if they interested.

Eric89GXL, 2015-11-09:
In `ext` we could first try to load the system version and fall back to our bundled version. Then the only thing the patch needs to do is exclude certain files from `mne/externals/`.

agramfort, 2015-11-09:
I would not change anything to master but apply patches during packaging
for the different distributions. As we do for debian

dengemann, 2015-11-09:
+100

Eric89GXL, 2016-02-03:
Closing as we've decided to use the patch-and-package method instead

closed 2016-02-03 by Eric89GXL