mne-tools/mne-python #2603: [py3] Brainvision tests fails due UnicodeDecodeError

ignatenkobrain, 2015-11-09:
I'm using last released version of MNE (probably you will thnk that it's because I unbundled libs -> no. this build with original release).
```python
======================================================================
ERROR: Test reading raw Brain Vision files
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/nose/case.py", line 198, in runTest
    self.test(*self.arg)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/tests/test_brainvision.py", line 35, in test_brainvision_data_filters
    preload=True)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/brainvision.py", line 511, in read_raw_brainvision
    response_trig_shift=response_trig_shift)
  File "<string>", line 2, in __init__
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/utils.py", line 549, in verbose
    return function(*args, **kwargs)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/brainvision.py", line 79, in __init__
    vhdr_fname, eog, misc, response_trig_shift, scale)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/brainvision.py", line 303, in _get_vhdr_info
    l = f.readline().strip()
  File "/usr/lib64/python3.4/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
nose.proxy.UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 638: ordinal not in range(128)
-------------------- >> begin captured stdout << ---------------------
Extracting parameters from /builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/tests/data/test_highpass.vhdr...
--------------------- >> end captured stdout << ----------------------
======================================================================
ERROR: Test reading raw Brain Vision files
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/nose/case.py", line 198, in runTest
    self.test(*self.arg)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/tests/test_brainvision.py", line 48, in test_brainvision_data
    raw_py = read_raw_brainvision(vhdr_path, montage, eog=eog, preload=True)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/brainvision.py", line 511, in read_raw_brainvision
    response_trig_shift=response_trig_shift)
  File "<string>", line 2, in __init__
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/utils.py", line 549, in verbose
    return function(*args, **kwargs)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/brainvision.py", line 79, in __init__
    vhdr_fname, eog, misc, response_trig_shift, scale)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/brainvision.py", line 303, in _get_vhdr_info
    l = f.readline().strip()
  File "/usr/lib64/python3.4/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
nose.proxy.UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 638: ordinal not in range(128)
-------------------- >> begin captured stdout << ---------------------
Extracting parameters from /builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/tests/data/test.vmrk...
Extracting parameters from /builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/tests/data/test.vhdr...
Extracting parameters from /builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/tests/data/test.vhdr...
--------------------- >> end captured stdout << ----------------------
======================================================================
ERROR: Test reading and modifying events
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/nose/case.py", line 198, in runTest
    self.test(*self.arg)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/tests/test_brainvision.py", line 92, in test_events
    raw = read_raw_brainvision(vhdr_path, eog=eog, preload=True)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/brainvision.py", line 511, in read_raw_brainvision
    response_trig_shift=response_trig_shift)
  File "<string>", line 2, in __init__
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/utils.py", line 549, in verbose
    return function(*args, **kwargs)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/brainvision.py", line 79, in __init__
    vhdr_fname, eog, misc, response_trig_shift, scale)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/brainvision.py", line 303, in _get_vhdr_info
    l = f.readline().strip()
  File "/usr/lib64/python3.4/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
nose.proxy.UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 638: ordinal not in range(128)
-------------------- >> begin captured stdout << ---------------------
Extracting parameters from /builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/tests/data/test.vhdr...
--------------------- >> end captured stdout << ----------------------
======================================================================
ERROR: Test writing raw eeg files when preload is False
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/nose/case.py", line 198, in runTest
    self.test(*self.arg)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/tests/test_brainvision.py", line 179, in test_read_segment
    raw1 = read_raw_brainvision(vhdr_path, eog=eog, preload=False)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/brainvision.py", line 511, in read_raw_brainvision
    response_trig_shift=response_trig_shift)
  File "<string>", line 2, in __init__
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/utils.py", line 549, in verbose
    return function(*args, **kwargs)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/brainvision.py", line 79, in __init__
    vhdr_fname, eog, misc, response_trig_shift, scale)
  File "/builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/brainvision.py", line 303, in _get_vhdr_info
    l = f.readline().strip()
  File "/usr/lib64/python3.4/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
nose.proxy.UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 638: ordinal not in range(128)
-------------------- >> begin captured stdout << ---------------------
Extracting parameters from /builddir/build/BUILD/python3-python-mne-0.10-1.fc24/mne/io/brainvision/tests/data/test.vhdr...
--------------------- >> end captured stdout << ----------------------
----------------------------------------------------------------------
```

Eric89GXL, 2015-11-09:
I think this is a bit of a misleading error. It's likely we have printed a non-ASCII character and `nose` captures it, and when it tries to print the actual error, it creates its own error. (I've seen this annoying issue before.) Can you try running those tests with the `-s` option, which will not capture stdout, thereby working around the silly `nose` bug? Then we can see what underlying `mne` errors you're getting.

ignatenkobrain, 2015-11-09:
```
+ nosetests-3.4 -v -s mne/io/brainvision/tests/test_brainvision.py
Test reading raw Brain Vision files ... Extracting parameters from /builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/tests/data/test_highpass.vhdr...
ERROR
Test reading raw Brain Vision files ... Extracting parameters from /builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/tests/data/test.vmrk...
Extracting parameters from /builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/tests/data/test.vhdr...
Extracting parameters from /builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/tests/data/test.vhdr...
ERROR
Test reading and modifying events ... Extracting parameters from /builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/tests/data/test.vhdr...
ERROR
Test writing raw eeg files when preload is False ... Extracting parameters from /builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/tests/data/test.vhdr...
ERROR

======================================================================
ERROR: Test reading raw Brain Vision files
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/nose/case.py", line 198, in runTest
    self.test(*self.arg)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/tests/test_brainvision.py", line 35, in test_brainvision_data_filters
    preload=True)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/brainvision.py", line 511, in read_raw_brainvision
    response_trig_shift=response_trig_shift)
  File "<decorator-gen-92>", line 2, in __init__
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/utils.py", line 549, in verbose
    return function(*args, **kwargs)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/brainvision.py", line 79, in __init__
    vhdr_fname, eog, misc, response_trig_shift, scale)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/brainvision.py", line 303, in _get_vhdr_info
    l = f.readline().strip()
  File "/usr/lib64/python3.4/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 638: ordinal not in range(128)

======================================================================
ERROR: Test reading raw Brain Vision files
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/nose/case.py", line 198, in runTest
    self.test(*self.arg)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/tests/test_brainvision.py", line 48, in test_brainvision_data
    raw_py = read_raw_brainvision(vhdr_path, montage, eog=eog, preload=True)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/brainvision.py", line 511, in read_raw_brainvision
    response_trig_shift=response_trig_shift)
  File "<decorator-gen-92>", line 2, in __init__
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/utils.py", line 549, in verbose
    return function(*args, **kwargs)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/brainvision.py", line 79, in __init__
    vhdr_fname, eog, misc, response_trig_shift, scale)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/brainvision.py", line 303, in _get_vhdr_info
    l = f.readline().strip()
  File "/usr/lib64/python3.4/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 638: ordinal not in range(128)

======================================================================
ERROR: Test reading and modifying events
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/nose/case.py", line 198, in runTest
    self.test(*self.arg)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/tests/test_brainvision.py", line 92, in test_events
    raw = read_raw_brainvision(vhdr_path, eog=eog, preload=True)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/brainvision.py", line 511, in read_raw_brainvision
    response_trig_shift=response_trig_shift)
  File "<decorator-gen-92>", line 2, in __init__
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/utils.py", line 549, in verbose
    return function(*args, **kwargs)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/brainvision.py", line 79, in __init__
    vhdr_fname, eog, misc, response_trig_shift, scale)
[... 7 lines cut ...]
ERROR: Test writing raw eeg files when preload is False
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/nose/case.py", line 198, in runTest
    self.test(*self.arg)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/tests/test_brainvision.py", line 179, in test_read_segment
    raw1 = read_raw_brainvision(vhdr_path, eog=eog, preload=False)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/brainvision.py", line 511, in read_raw_brainvision
    response_trig_shift=response_trig_shift)
  File "<decorator-gen-92>", line 2, in __init__
[... cut ...]
    vhdr_fname, eog, misc, response_trig_shift, scale)
  File "/builddir/build/BUILD/python-mne-0.10/python3/mne/io/brainvision/brainvision.py", line 303, in _get_vhdr_info
    l = f.readline().strip()
  File "/usr/lib64/python3.4/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 638: ordinal not in range(128)

----------------------------------------------------------------------
Ran 4 tests in 0.008s

FAILED (errors=4)
```

ignatenkobrain, 2015-11-09:
looks like it still prints nothing

agramfort, 2015-11-09:
test.vhdr contains utf8 characters such as µ

what I don't understand is that it works with travis running py3

Eric89GXL, 2015-11-09:
It is confusing to me, too. It can probably be fixed by changing the `open` to use binary mode and decoding. I'll try that and open a PR if it works.

Eric89GXL, 2015-11-09:
@agramfort is it worth making Travis run Py35 while I'm at it?

agramfort, 2015-11-09:
Yes please

PR #2605 (2015-11-09): FIX: Fixes for Py35
Closes #2603.

@ignatenkobrain can you see if this fixes your issue?
closed 2015-11-09 by Eric89GXL