mne-tools/mne-python #2608: BUG: ica plot_sources udpated exclude if exclude is passed and a raw object

dengemann, 2015-11-11:
the interactive machinery we added will have weird side effects ... 

```Python
ica.plot_sources(raw, ecg_inds, exclude=ecg_inds,
                            title=title % ('components', 'ecg'), show=show)
```

A call like this will update ica.exclude -- not what it is supposed to do.

maybe you can give me a hand @jaeilepp

dengemann, 2015-11-11:
same is apparently true for ``ica.plot_components``

dengemann, 2015-11-11:
@jaeilepp I figured that it happens when I save the figure to the report, suggesting that exclude is updated on closing the figure. Does that ring a bell?

jaeilepp, 2015-11-11:
Yeah, it does that. I guess we could add a flag that disables the interactive features.

jaeilepp, 2015-11-12:
I don't think ``plot_components`` modifies the exclude.

dengemann, 2015-11-12:
>  don't think plot_components modifies the exclude.

In my test I had a different impression but I agree, it would be surprising based on the code.

>  I guess we could add a flag that disables the interactive features.

Or make sure that passing exclude values is not updating anything unless the user wants it (clicks).

jaeilepp, 2015-11-13:
>      Or make sure that passing exclude values is not updating anything unless the user wants it (clicks).

There is just this problem that if the components marked with param ``exclude`` are plotted red and the clicked components are red, they look the same for the user. One option is to use different colors for them, but that would probably be even more confusing. I'd go with making sure that the documentation is crystal clear on the behavior of exclude parameter.

dengemann, 2015-11-13:
@jaeilepp the problem is that the API has silentnly changed, in fact we broke some code this way. The paprameter is only for plotting not for updating the ICA components. We need to come up with a solution here. Can't we just update the components only if the user has actually clicked and this clicking actually results in a different set of components? For colors #2607 should help, although I did not implement it for the raw plot.

larsoner, 2019-08-20:
@drammock recently fixed this

closed 2019-08-20 by larsoner