mne-tools/mne-python #2613: BrainVision Analyser data format version 2

iikun, 2015-11-13:
Hello, 
I have some issues with importing Brainvision Files. When i do some preprocessing in Brainvision Analyser and then export the data, it exports data  headers in "Brain Vision Data Exchange Header File Version 2.0" format. It is slightly different from that, which is now implemented in MNE, so I had to brutally hack existing function to read my files. Unfortunately every update breaks my code and I don't feel up to the task of officially maintaining it on github. 

So I thought that it would be great if someone could review my code and maybe incorporate it into the MNE, you can find those dirty hacks here: http://pastebin.com/BbgsP4Wv

And some headers here:
[ICA_OBS_MRC_LTMAM2_01_p1.vhdr.txt](https://github.com/mne-tools/mne-python/files/34092/ICA_OBS_MRC_LTMAM2_01_p1.vhdr.txt)
[ICA_OBS_MRC_LTMAM2_01_p1.vmrk.txt](https://github.com/mne-tools/mne-python/files/34093/ICA_OBS_MRC_LTMAM2_01_p1.vmrk.txt)

agramfort, 2015-11-13:
the good approach is to send us the patch as a PR on github so we can integrate it and you don't have to maintain it.

agramfort, 2015-11-13:
please give it a try. And ask if you need help.

iikun, 2015-11-13:
Should I fork and do a PULL request on 0.9 branch, or should I do it on master?

agramfort, 2015-11-13:
do it on master and we can backport to 0.10 eventually.

mdovgialo, 2015-11-13:
I've created pull request.
I don't think code is too clean, I just created mne.io.read_raw_brainvision2 function for those headers. It is usable, but can do more. Header v2.0 saves electrode locations, but I never realized support for reading them.

teonlamont, 2015-11-13:
@mdovgialo what is the difference between the two functions? I would prefer to keep it one function that checks for the version to adapt for the different versions. It would simplify the code and we would be able to see what the differences are for the different versions too

jona-sassenhagen, 2015-11-13:
@teonlamont didn't you (or possibly I) fix some stuff in .9 that made reading of bv2 files possible?
mdovgialo have you tried it with a recent version?

mdovgialo, 2015-11-13:
I've done some checking. I remember that I had to change quite some stuff for it to work, in mne 0.8, mostly in marker - reading code. After upgrade to 0.9 my hacks didn't work, so I started over and 
@jona-sassenhagen you are right, there were fixes, and function in mine pull request is pretty much the same as in mne 0.9, only asserts are changed to "Version 2.0", and some units multipliers changed and I added cases for reading markers tagged as "Pulse Artifact" and "Bad Interval" (I needed them for my code). 

But the newest version was changed a lot, and I don't know if just changing asserts will work now.

teonlamont, 2015-11-13:
@mdovgialo sorry, but could you try using the v0.10 to see where the problem is now.

mdovgialo, 2015-11-13:
on 0.10 I get the same assertion error
/usr/local/lib/python2.7/dist-packages/mne/io/brainvision/brainvision.pyc in _get_vhdr_info(vhdr_fname, eog, misc, response_trig_shift, scale)
    302         # extract the first section to resemble a cfg
    303         l = f.readline().strip()
--> 304         assert l == 'Brain Vision Data Exchange Header File Version 1.0'
    305         settings = f.read()
    306

AssertionError:

In [71]:

It looks like it was ready for use in 0.9, but assertions for v2.0 headers where not added.

mdovgialo, 2015-11-15:
@teonlamont 
Sharing the v2.0 files:
http://www.fuw.edu.pl/~mdovgialo/ICA_OBS_MRC_LTMAM2_01_p1.vhdr
http://www.fuw.edu.pl/~mdovgialo/ICA_OBS_MRC_LTMAM2_01_p1.vmrk
http://www.fuw.edu.pl/~mdovgialo/ICA_OBS_MRC_LTMAM2_01_p1.dat

PR #2634 (2015-12-12): [MRG] ENH Add support for Brainvision V2.0
closes #2613.
Taking over #2614.

This is WIP. It provides reading support for BV 2.0.

Things to be addressed: BV optionally has the channel locations listed in the vhdr file. Should we have  this supported in the montage module with a custom file format option `*.vhdr`?
closed 2015-12-12 by Eric89GXL