mne-tools/mne-python #2622: Thoughts on xray data structures?

choldgraf, 2015-11-17:
Have you guys looked into xray at all? It seems pretty nifty, though probably too early to say if it'll be widely adopted or not:

http://xray.readthedocs.org/en/stable/data-structures.html

Right now we have methods like to_dataframe, but it sounds like to_dataarray would make more sense for something like epochs data. Just bringing this up to see what people think about it.

agramfort, 2015-11-17:
write a blog post to convince us? :)

seriously I've used xray in the past and it's cool but I am not sure we are
willing to add more code to mne for this. We moved back on pandas...

my 2c

jona-sassenhagen, 2015-11-17:
I think it'd be really cool cause I'm an idiot who always forgets what the dimensions are and it would be simpler if I could do something like .mean("time") or .mean("sensors") instead.

jona-sassenhagen, 2015-11-17:
Maybe minimally there could be an xray equivalent of `.get_data` (which returns pandas data frames).

dengemann, 2015-11-17:
> So there's already a to_dataframe method with several objects, and I think
> the trick with using xray or not is that you can do lots of the same kind
> of functionality with DataFrames already.
>
> E.g., since each datapoint is labeled in a DataFrame, you can just say
> "apply this function across a given level of a multi-index", and that will
> basically do the same thing as DataArray.apply_function_over_axis (don't
> know what the actual method is called).
>
> Here's a function I wrote to do this. It's not tested or very robust, but
> maybe it's useful?
>
> def apply_across_df_level(df, levels, func=np.mean):
>     """Average out a level from a dataframe.    This is a convenience function to quickly average away a set    of levels in a dataframe. The dataframe will be grouped by all    other levels, and then averaged.    Parameters    ----------    df : DataFrame        The dataframe in question    levels : list of strings        The names of the levels to average out    Returns    -------    df : DataFrame        The input dataframe with levels averaged out    """
>     if not isinstance(levels, (list, tuple)):
>         raise ValueError('level names must be in a list')
>     count_levels = filter(lambda a: a not in levels, df.index.names)
>     print('Retaining levels: {0}'.format(count_levels))
>     df = df.groupby(level=count_levels)
>     df = df.agg(func)
>     return df
>
> So the functionality is already there, it's just a bit more clunky with
> something like a DataFrame.

Sorry to crash the party. Do we really need this here in MNE? A nice blog
post would do it too. It's Python, everyone can assemble his pipeline by
putting together elements from different libraries. We already have Pandas.
And we had enough effort maintaining it so far.
If you want more sophisticated labled access to data let's maybe improve
our API. A core task that I see there is an internal abstraction that
handles time and frequency. E.g. it's confusing to abuse .times for
frequencies all over the place. We have enough open tasks on this end.

choldgraf, 2015-11-17:
Agreed that there are bigger fish to fry first. I think that having an EpochTFR-style object would be useful just because there are functions that assume the kind of data that an EpochTFR object would have, whereas the functions for decomposing a signal into a TFR tend to automatically average across trials, so it's a bit clunky in that sense. Maybe one day xray will be fully-featured enough to be worth implementing as a basic component of a neural analysis package, but probably that day is not today :P

choldgraf, 2015-11-20:
I'm gonna close this so that it's not cluttering up the issues, but maybe it can pop back up in the future :)

closed 2015-11-20 by choldgraf

jona-sassenhagen, 2015-11-20:
>I think that having an EpochTFR-style object would be useful just because there are functions that assume the kind of data that an EpochTFR object would have, whereas the functions for decomposing a signal into a TFR tend to automatically average across trials
FWIW @choldgraf I want to extend linear_regression_raw to the time-frequency domain eventually.

choldgraf, 2015-11-20:
Cool - I think this would be great as well. Check out #2024 where we discussed this a bit a while ago, and I believe I posted some code for something like an "EpochTFR" analysis. I've been meaning to take a look at the linear_regression code too, as all of my work is using regression for encoding models in ECoG :)

jona-sassenhagen, 2015-11-20:
Note an EpochTFR thing would be, if at all, for linear_regression. linear_regression_raw circumvents epochs. linear_regression_raw with a TF option would directly return AverageTFR objects per condition estimated from a continuous (raw?) file.

jona-sassenhagen, 2015-11-20:
EpochTFR would still be cool for example if we want to do something like decoding in the time-frequency domain.

choldgraf, 2015-11-20:
Agreed - I just meant that as an aside, not as directly related to EpochTFR. The challenge with the full TFR decomposition is just that it tends to combinatorially explode analysis time, but would be very cool to be able to keep that information for decoding purposes etc. One issue I've run into with this object is the I/O can be expensive depending on how much precision you want in the numbers. If you store them as higher-bit numbers, then the size of files can get up to like 5-6GB and it was making our sysadmin unhappy with me...

jona-sassenhagen, 2015-11-20:
Ah right- I'm thinking 30 channel EEG from 1 to 30 Hz here :)